repetier/RepetierFirmware #744: Feature request V2: Add support for Teensy 3.6

boelle, 2018-01-21:
I wonder if this would be an easy one to add?

Seems like a fast board, just have to make an Ramps like board for it. Shouldnot take long since much can be copied from the eagle files

repetier, 2018-01-21:
It mainly depends on teensy 3.5 or 3.6. That is a different processor then the due so the HAL part will not be compatible. If the arduino are available for it (there are some but not sure for which processors) it would be just adding another hardware package. Maybe it is easier with start otto which might have less compatibility issues and also has 180MHz if it will appear for sale.

boelle, 2018-01-21:
what will appear for sale? you lost me a bit :-D

repetier, 2018-01-21:
Arduino Star Otto. Also there are some unboxing videos it seems impossible to order them. At least last time I checked.

boelle, 2018-01-21:
did a google for it, but like you say hard to find

at least the teensy 3.6 is easy to get

closed 2019-03-19 by boelle