repetier/RepetierFirmware #745: Errors when using MAX6675 in 1.0.1

overdrive2max, 2018-01-26:
Hi
In version 1.0.1 downloaded via the configuration tool I have found the following errors:

In Extruder.cpp the following errors are present:
Line 2171: "}"
line 2557 and 2559 there are 2 incorrect statements
        max6675_temp[idx] = HAL::spiReceive(0);
Must be 
        max6675_temp[idx] = HAL::spiReceive();

repetier, 2018-01-27:
Actually this was only an error in HAL version of due not accepting an argument set to default value like the AVR version.

I assume you had this on a due on a due based system.

Github version is fixed and online will follow soon.

closed 2018-01-27 by repetier

overdrive2max, 2018-01-27:
Jip, Due based board. Was the omitted "}" also fixed?

repetier, 2018-01-27:
Did not notice this part. I snow also fixed. Thank you.

overdrive2max, 2018-01-28:
Pleasure.