repetier/RepetierFirmware #747: Front Panel infinite loop on distortion correction

Masterkookus, 2018-01-28:
Added line at 2599 of ui.cpp to fix the issue, not sure if this is isolated to me or if it should be added to code:
uid.popMenu(false);

repetier, 2018-01-28:
Can you show more of the code. I guess you mean this function:

#if DISTORTION_CORRECTION
            case UI_ACTION_MEASURE_DISTORTION2:
                uid.pushMenu(&ui_msg_calibrating_bed, true);
                Printer::measureDistortion();
                uid.popMenu(true);
                break;
#endif

which starts in current version at line 2597 to 2599 is the push method. So is this a different version or did you add it before the push? That would reduce menu stack by one level when you return from distortion correction. Should only a problem if the maximum menu depth gets exceeded.

Masterkookus, 2018-01-28:
Yes, I added it as below, without it I just kept getting returned to the bed calibrating ok screen.

#if DISTORTION_CORRECTION
            case UI_ACTION_MEASURE_DISTORTION2:
                uid.popMenu(false);
                uid.pushMenu(&ui_msg_calibrating_bed, true);
                Printer::measureDistortion();
                uid.popMenu(true);
                break;
#endif