repetier/RepetierFirmware #748: Activating z-probe would lead to forbidden xy position

thiagopeixoto16, 2018-02-02:
Hello,

I have a servo mounted Z probe acting as Z_min (same pin) on a Cartesian printer,
When I run Z homing (G28 Z0 or G28) it is **not** going to ZHOME_X_POS and ZHOME_Y_POS.
Due that I'm getting the error "_Activating z-probe would lead to forbidden xy position_" from host.

This is my parameters:
#define HOMING_ORDER HOME_ORDER_XYZ
#define ZHOME_X_POS 5 //IGNORE_COORDINATE
#define ZHOME_Y_POS 30 //IGNORE_COORDINATE
#define Z_PROBE_X_OFFSET 5 (in the EEPROM)
#define Z_PROBE_Y_OFFSET 25 (in the EEPROM)

Using 1.0 master version.
[Here](https://www.dropbox.com/s/l18hhouboszckvh/EEPROM.epr?dl=0) you have my EEPROM

Thanks,

repetier, 2018-02-03:
You need a homing order with preheat to use the zhome positions. Set preheat temperature to 0 to skip that part of the function.

thiagopeixoto16, 2018-02-05:
I tried:
#define Z_PROBE_REQUIRES_HEATING 1
#define Z_PROBE_MIN_TEMPERATURE 0 (I tried "1" also)
#define ZHOME_MIN_TEMPERATURE 1

Same issue...

repetier, 2018-02-06:
#define HOMING_ORDER HOME_ORDER_XYZ
defines the homing order which I told you to change to a preheat version for homing coord inclusion.

thiagopeixoto16, 2018-02-07:
Alright, I had to go to online interface to get what is the matter.
In order to facilitate another people's life, the solution for this issue is:
**Change HOMING_ORDER to HOME_ORDER_XYTZ to enable the ZHOME_X_POS/ZHOME_Y_POS feature.**

Thanks @repetier !

closed 2018-02-07 by thiagopeixoto16