repetier/RepetierFirmware #752: Distortion correction: From where "Reference Z for measurement" is taken from [doubt]

thiagopeixoto16, 2018-02-12:
In my case is is always 20.000 no matter what I change...
I want to understand if it is a fixed value or not.

Another side questions is that it is reading always +2.7ish since the first test point:
Z-probe:22.793 X:7.00 Y:200.00

It changes the compensation of z-probe height, and that's ok if I fix it after the G33, but again I just want to understand better.

Thanks again!

repetier, 2018-02-13:
Start distance is z probe bed distance + z probe height (+0.1 maybe). Does that make sense with your values in eeprom?

thiagopeixoto16, 2018-02-13:
Yes, it is the formula you pointed, Not exatly but I got it. Thanks!
I have another problem with the distortion, but I'll open in another issue.

closed 2018-02-13 by thiagopeixoto16