repetier/RepetierFirmware #753: Distortion correction matrix too high

thiagopeixoto16, 2018-02-13:
Hi,

I've followed the video and the procedure on [documentation page](https://www.repetier.com/documentation/repetier-firmware/z-probing/), however the resulting matrix has high values like that:
-11058, -11714, -11548, -11566, -11260, -11672
-11442, -12070, -12186, -12002, -11532, -11732
-11558, -12192, -12338, -12166, -12010, -11874
-11768, -12376, -12376, -12146, -11730, -11780
-11792, -12274, -12236, -11878, -11612, -11478
-11804, -12076, -11772, -11498, -11090, -11006

The Repetier Host tool shows that it is ok, check the before and after:
**Before:**

```
X:		7,00	45,60	84,20	122,80	161,40	200,00
y:200,00	22,81	22,97	22,93	22,94	22,87	22,97
y:165,40	22,88	23,05	23,09	23,06	22,96	23,00
y:130,80	22,90	23,09	23,13	23,10	23,07	23,04
y:96,20		22,94	23,11	23,13	23,08	22,99	23,02
y:61,60		22,94	23,08	23,09	22,99	22,95	22,95
y:27,00		22,92	23,03	22,98	22,93	22,84	22,82
```
**After:**
```
X:		7,00	45,60	84,20	122,80	161,40	200,00
y:200,00	20,00	19,99	19,98	19,99	19,99	19,97
y:165,40	19,97	19,99	19,98	19,99	19,98	19,97
y:130,80	19,99	19,97	19,98	19,99	19,97	19,97
y:96,20		20,00	19,98	19,99	19,99	19,96	19,96
y:61,60		20,00	19,99	19,99	19,98	19,98	19,98
y:27,00		19,98	19,99	20,00	20,01	20,00	19,99
```
The problem is: when I try to print (ou move the head) the first layer height is crazy in some points of the bed. Specially in the X200Y0 to X200Y200. What's wrong?

Below the steps taken:

> 1. Clear matrix M322 S2
> 2. Home all G28
> 3. Auto bed correction G32 S2 
> 4. G33 R0 clean distortion
> 5. home all G28 
> 6. G1 X100 Y100 Z30 go to center to avoid "Forbidden position"
> 7. G33

The results I have for G33 are:
```
Z-probe:24.564 X:7.00 Y:200.00
Z-probe:24.728 X:45.60 Y:200.00
Z-probe:24.687 X:84.20 Y:200.00
Z-probe:24.691 X:122.80 Y:200.00
Z-probe:24.615 X:161.40 Y:200.00
Z-probe:24.718 X:200.00 Y:200.00
Z-probe:24.660 X:7.00 Y:165.40
Z-probe:24.817 X:45.60 Y:165.40
Z-probe:24.846 X:84.20 Y:165.40
Z-probe:24.800 X:122.80 Y:165.40
Z-probe:24.683 X:161.40 Y:165.40
Z-probe:24.733 X:200.00 Y:165.40
Z-probe:24.689 X:7.00 Y:130.80
Z-probe:24.848 X:45.60 Y:130.80
Z-probe:24.884 X:84.20 Y:130.80
Z-probe:24.841 X:122.80 Y:130.80
Z-probe:24.802 X:161.40 Y:130.80
Z-probe:24.768 X:200.00 Y:130.80
Z-probe:24.742 X:7.00 Y:96.20
Z-probe:24.894 X:45.60 Y:96.20
Z-probe:24.894 X:84.20 Y:96.20
Z-probe:24.836 X:122.80 Y:96.20
Z-probe:24.732 X:161.40 Y:96.20
Z-probe:24.745 X:200.00 Y:96.20
Z-probe:24.748 X:7.00 Y:61.60
Z-probe:24.868 X:45.60 Y:61.60
Z-probe:24.859 X:84.20 Y:61.60
Z-probe:24.769 X:122.80 Y:61.60
Z-probe:24.703 X:161.40 Y:61.60
Z-probe:24.669 X:200.00 Y:61.60
Z-probe:24.751 X:7.00 Y:27.00
Z-probe:24.819 X:45.60 Y:27.00
Z-probe:24.743 X:84.20 Y:27.00
Z-probe:24.674 X:122.80 Y:27.00
Z-probe:24.572 X:161.40 Y:27.00
Z-probe:24.551 X:200.00 Y:27.00
```

thiagopeixoto16, 2018-02-13:
Sure, here find the missing information:
> Repetier 1.0.0 Master
> Z-probe is the Z-min
> Cartesian printer

**G1 X100 Y100 Z10 results:**
> Same odd G33 L0 like below:

**G33 L0 results**
```
13:31:06.164 : G33   X7.00 Y27.00 Z-2.951
13:31:06.166 : G33 X7.00 Y61.60 Z-2.948
13:31:06.169 : G33 X7.00 Y96.20 Z-2.942
13:31:06.172 : G33 X7.00 Y130.80 Z-2.890
13:31:06.175 : G33 X7.00 Y165.40 Z-2.861
13:31:06.177 : G33 X7.00 Y200.00 Z-2.765
13:31:06.178 : G33 X45.60 Y27.00 Z-3.019
13:31:06.180 : G33 X45.60 Y61.60 Z-3.069
13:31:06.183 : G33 X45.60 Y96.20 Z-3.094
13:31:06.186 : G33 X45.60 Y130.80 Z-3.048
13:31:06.188 : G33 X45.60 Y165.40 Z-3.018
13:31:06.191 : G33 X45.60 Y200.00 Z-2.929
13:31:06.194 : G33 X84.20 Y27.00 Z-2.943
13:31:06.194 : G33 X84.20 Y61.60 Z-3.059
13:31:06.196 : G33 X84.20 Y96.20 Z-3.094
13:31:06.199 : G33 X84.20 Y130.80 Z-3.085
13:31:06.202 : G33 X84.20 Y165.40 Z-3.047
13:31:06.205 : G33 X84.20 Y200.00 Z-2.887
13:31:06.208 : G33 X122.80 Y27.00 Z-2.875
13:31:06.211 : G33 X122.80 Y61.60 Z-2.970
13:31:06.212 : G33 X122.80 Y96.20 Z-3.037
13:31:06.214 : G33 X122.80 Y130.80   Z-3.042
13:31:06.217 : G33 X122.80 Y165.40   Z-3.001
13:31:06.220 : G33 X122.80 Y200.00   Z-2.892
13:31:06.223 : G33 X161.40 Y27.00 Z-2.773
13:31:06.226 : G33 X161.40 Y61.60 Z-2.903
13:31:06.229 : G33 X161.40 Y96.20 Z-2.933
13:31:06.232 : G33 X161.40 Y130.80   Z-3.003
13:31:06.235 : G33 X161.40 Y165.40   Z-2.883
13:31:06.238 : G33 X161.40 Y200.00   Z-2.815
13:31:06.238 : G33 X200.00 Y27.00 Z-2.752
13:31:06.241 : G33 X200.00 Y61.60 Z-2.870
13:31:06.245 : G33 X200.00 Y96.20 Z-2.945
13:31:06.248 : G33 X200.00 Y130.80   Z-2.969
13:31:06.251 : G33 X200.00 Y165.40   Z-2.933
13:31:06.253 : G33 X200.00 Y200.00   Z-2.918
```
Working now on **G32 S2 and go to z=0** testing

thiagopeixoto16, 2018-02-13:
I think you found something:
"G32 S2 and go to z=0" test is actually positioning the nozzle -2.9mm from the bed surface.

I could change the z-probe height to fix that, but it will make Z homing useless.
Anyway I tried, and the G33 L0 results (and the attempt to move around the bed) are the same:
```
15:33:28.605 : G33 X7.00 Y27.00 Z-2.941
15:33:28.608 : G33 X7.00 Y61.60 Z-2.942
15:33:28.610 : G33 X7.00 Y96.20 Z-2.939
15:33:28.613 : G33 X7.00 Y130.80 Z-2.890
15:33:28.616 : G33 X7.00 Y165.40 Z-2.867
15:33:28.619 : G33 X7.00 Y200.00 Z-2.782
15:33:28.619 : G33 X45.60 Y27.00 Z-3.037
15:33:28.621 : G33 X45.60 Y61.60 Z-3.087
15:33:28.624 : G33 X45.60 Y96.20 Z-3.113
15:33:28.627 : G33 X45.60 Y130.80 Z-3.100
15:33:28.630 : G33 X45.60 Y165.40 Z-3.040
15:33:28.632 : G33 X45.60 Y200.00 Z-2.944
15:33:28.635 : G33 X84.20 Y27.00 Z-2.958
15:33:28.635 : G33 X84.20 Y61.60 Z-3.077
15:33:28.638 : G33 X84.20 Y96.20 Z-3.116
15:33:28.641 : G33 X84.20 Y130.80 Z-3.107
15:33:28.643 : G33 X84.20 Y165.40 Z-3.076
15:33:28.646 : G33 X84.20 Y200.00 Z-2.913
15:33:28.649 : G33 X122.80 Y27.00 Z-2.907
15:33:28.653 : G33 X122.80 Y61.60 Z-3.005
15:33:28.653 : G33 X122.80 Y96.20 Z-3.076
15:33:28.656 : G33 X122.80 Y130.80 Z-3.078
15:33:28.659 : G33 X122.80 Y165.40 Z-3.036
15:33:28.662 : G33 X122.80 Y200.00 Z-2.923
15:33:28.665 : G33 X161.40 Y27.00 Z-2.805
15:33:28.668 : G33 X161.40 Y61.60 Z-2.936
15:33:28.671 : G33 X161.40 Y96.20 Z-2.970
15:33:28.674 : G33 X161.40 Y130.80 Z-3.040
15:33:28.677 : G33 X161.40 Y165.40 Z-2.933
15:33:28.680 : G33 X161.40 Y200.00 Z-2.837
15:33:28.680 : G33 X200.00 Y27.00 Z-2.804
15:33:28.683 : G33 X200.00 Y61.60 Z-2.922
15:33:28.686 : G33 X200.00 Y96.20 Z-3.002
15:33:28.689 : G33 X200.00 Y130.80 Z-3.034
15:33:28.692 : G33 X200.00 Y165.40 Z-2.982
15:33:28.695 : G33 X200.00 Y200.00 Z-2.958
```

repetier, 2018-02-13:
Ok, z probe height is wrong by 2.9mm.

Homing should work with correct z height. Make sure bed coating is set 0 and that z min pin is the same a z probe pin, homing order with preheat function selected so you can define xy for z probing. Set preheat temp to 0 if not needed.

thiagopeixoto16, 2018-02-13:
Yes, after G33 mapped Z0 gets corrected. 
- I don't activated bead coating
- Z probe and z min are on the same pin already
- Homing order is like that (based on you last orientation - tks)

The issue persists, completely crazy position around the bed...

What I'm doing now is:
Getting all Z readings provided by G33 L0, copy and paste to excel and normalizing it like:
1. Finding the middle measurement between the max and min Z in this list, called M;
2. Subtracting M from every G33 L0 value;
3. Generating a new command list with this values
4. Resetting (G33 R0) and sending all values to firmware via GCODE:
I have something like this:
```
G33 X7.00 Y27.00   Z-0.028
G33 X7.00 Y61.60 Z-0.025
G33 X7.00 Y96.20 Z-0.019
G33 X7.00 Y130.80 Z0.033
G33 X7.00 Y165.40 Z0.062
G33 X7.00 Y200.00 Z0.158
G33 X45.60 Y27.00 Z-0.096
G33 X45.60 Y61.60 Z-0.146
G33 X45.60 Y96.20 Z-0.171
G33 X45.60 Y130.80 Z-0.125
G33 X45.60 Y165.40 Z-0.095
G33 X45.60 Y200.00 Z-0.006
G33 X84.20 Y27.00 Z-0.02
G33 X84.20 Y61.60 Z-0.136
G33 X84.20 Y96.20 Z-0.171
G33 X84.20 Y130.80 Z-0.162
G33 X84.20 Y165.40 Z-0.124
G33 X84.20 Y200.00 Z0.036
G33 X122.80 Y27.00 Z0.048
G33 X122.80 Y61.60 Z-0.047
G33 X122.80 Y96.20 Z-0.114
G33 X122.80 Y130.80  Z-0.119
G33 X122.80 Y165.40  Z-0.078
G33 X122.80 Y200.00  Z0.031
G33 X161.40 Y27.00 Z0.15
G33 X161.40 Y61.60 Z0.02
G33 X161.40 Y96.20 Z-0.01
G33 X161.40 Y130.80  Z-0.08
G33 X161.40 Y165.40  Z0.04
G33 X161.40 Y200.00  Z0.108
G33 X200.00 Y27.00 Z0.171
G33 X200.00 Y61.60 Z0.053
G33 X200.00 Y96.20 Z-0.022
G33 X200.00 Y130.80  Z-0.046
G33 X200.00 Y165.40  Z-0.01
G33 X200.00 Y200.00  Z0.005
```
I did it and tested: The craziness movements are gone but my bed bumps weren't fixed.

I'm now doing a test with no distortion (R0) and just to have a fixed comparison. 
I'm suspecting that the number signals are inverted on the above excel list.

thiagopeixoto16, 2018-02-14:
Update: It is working with this last workaround. The bumps were improved but not completely solved.
I'll round the values to two decimals just to check if a little bit aggressive values do the trick.
More updates tomorrow.

thiagopeixoto16, 2018-02-14:
Yes, it is definitely working but only if I manually (excel) fix the readings...
Any light on this?

repetier, 2018-02-15:
Note that approximation on cartesian is not the best. Moves are split in 10mm segments for correction at end points. So if you are 5mm left and right to a peak you will not go up to the peak but go through it. With small differences this is no real problem. In V2 firmware there will be more updates like on deltas now.

What did you fix in excel? If you need to fix something it is still not perfect and should be adjustable by parameter, but you did not provide all informations. Just that it is now better, but not why and new results.

thiagopeixoto16, 2018-02-15:
1. If I test using circles (or curves) will give more segments? 
2. Check the items 1 to 4 I've explained on the previous post. That contains the steps I did and also the resulting list.
3. Here find below the visual results. 
First image is about **no correction** at all (G33 R0)
[image]

Second image: Distortion correction ON using the values I have corrected on excel (list on previous post):
[image]

thiagopeixoto16, 2018-02-16:
@repetier Do you have all information now? May I help with something else?

thiagopeixoto16, 2018-04-19:
and....

thiagopeixoto16, 2018-04-20:
It still happen.
Yes, it is. I have a perfect first layer despite the bumps.

I found no "percent" value. This is what I have on distortion:
```
#define DISTORTION_CORRECTION         1
#define DISTORTION_CORRECTION_POINTS  6
#define DISTORTION_LIMIT_TO 0
#define DISTORTION_CORRECTION_R       0
#define DISTORTION_XMIN 10
#define DISTORTION_YMIN 27
#define DISTORTION_XMAX 200
#define DISTORTION_YMAX 200
#define DISTORTION_PERMANENT          1
#define DISTORTION_UPDATE_FREQUENCY   15
#define DISTORTION_START_DEGRADE 0.3
#define DISTORTION_END_HEIGHT 1.5
#define DISTORTION_EXTRAPOLATE_CORNERS 0
```

repetier, 2018-04-20:
#define DISTORTION_START_DEGRADE 0.3
#define DISTORTION_END_HEIGHT 1.5
are the 2 heights. So until 0.3mm it corrects 100% and at 1.5mm it stops with correction.