repetier/RepetierFirmware #754: possible bug in G32 in v1.0.1

NojbyMichal, 2018-02-15:
Hi, i am now testing repetier fw version 1.0.1 on my delta, but found possible bug in G32 S2 command. Before updating to 1.0.1 i used 0.92.9 (or 0.92.8) on the same delta printer, where G32 works, so I think that it could't be a mechanical or eletrical problem .. like small current into motor coils ...

Description of the problem: When I start G32 S2, the printer make home and after that the head with probe is moving down, it suddenly stops. All three motors stalls and in repetier-host I see that the printer resets itself. If needed I can add more information, repetier-host snapshot, record video..., don't hesitate to ask. 

Thank  you for your help. Michal

repetier, 2018-02-15:
Hi,

a typical error when switching is to use same number of subsegments per line. 1.0.1 uses more variables and has therefore less ram. So my guess is that on startup it shows less then 900 free bytes and that causes the firmware crash while homing.

Another reason would be if your speed to fast eating up all cpu time. That could cause watchdog to trigger.

NojbyMichal, 2018-02-15:
Thats crash isn't while homing, it is after homing. 

1) head is going to home position -> everything is OK
2) make a home (tact on all endstops) -> OK
3) head is moving down to measure the bed -> while it is moving  it suddenly stall  (ERROR)

My Max travel speed is actually 150mm/s.

In the evening I can make a test, lowering down the moving speed and lowering down the movement segments to a half that I have. I report the results.

repetier, 2018-02-15:
If printer resets anything can happen. So please check memory issue first to see if it then doesn't reset.