repetier/RepetierFirmware #756: Code changes for RuRAMPS4Due board

dell9, 2018-02-27:
Hello! 

Found the following problems with code for RuRAMPS4Due board in the v1.0.1:
**DisplayList.h** line 301:
`#elif MOTHERBOARD == 413 // RURAMPS4D`
should be changed to
`#elif MOTHERBOARD == 414 // RURAMPS4D`

**pins.h** line 1173:
```
// Note that on the Due pin A0 on the board is channel 2 on the ARM chip
#define HEATER_0_PIN     13
// Due analog pin #54
#define TEMP_0_PIN       7 
#define HEATER_1_PIN     7
#define TEMP_1_PIN       6 
// Due analog pin #58
#define HEATER_2_PIN     12
// Due analog pin #55
#define TEMP_2_PIN       5 
#define HEATER_3_PIN     11
// Due analog pin #56
#define TEMP_3_PIN       4 
// Due analog pin #57
//#define TEMP_4_PIN       4 
```
should be changed to
```
// Note that on the Due pin A0 on the board is channel 2 on the ARM chip
#define HEATER_0_PIN     13
// Due analog pin #54
#define TEMP_0_PIN       7 
#define HEATER_1_PIN     7
// Due analog pin #58
#define TEMP_1_PIN       3 
#define HEATER_2_PIN     12
// Due analog pin #55
#define TEMP_2_PIN       6 
#define HEATER_3_PIN     11
// Due analog pin #56
#define TEMP_3_PIN       5 
// Due analog pin #57
#define TEMP_4_PIN       4
```

dell9, 2018-03-17:
Another problem. The following code in ui.h breaks SD detect for RuRAMPS4Due board and RepRapDiscount Full Graphic Smart Controller:

```
#if FEATURE_CONTROLLER == CONTROLLER_SMARTRAMPS || FEATURE_CONTROLLER == CONTROLLER_GADGETS3D_SHIELD || FEATURE_CONTROLLER == CONTROLLER_BAM_DICE_DUE || (FEATURE_CONTROLLER == CONTROLLER_REPRAPDISCOUNT_GLCD && MOTHERBOARD != CONTROLLER_FELIX_DUE  && MOTHERBOARD != 101)
#undef SDCARDDETECT
#if MOTHERBOARD == 37
#define SDCARDDETECT ORIG_SDCARDDETECT
#else
#define SDCARDDETECT 49
#endif
#undef SDCARDDETECTINVERTED
#define SDCARDDETECTINVERTED 0
#undef SDSUPPORT
#define SDSUPPORT 1
#endif
```
You need to add exclusion for MOTHERBOARD  = 414 since it uses SDCARDDETECT  = 51.  Code can be the following:

```
#if MOTHERBOARD == 37 || MOTHERBOARD  == 414
#define SDCARDDETECT ORIG_SDCARDDETECT
```

repetier, 2018-03-18:
Looking at http://3dtoday.ru/blogs/svs0724/ruramps-pinout-the-pinout-of-the-connectors-on-the-board/ I saw that T4 was in deed not listed. But there is no need to change the numbering like you did. It is more logical to have TEMP_4_PIN connected to T4 pin.

Display changes have been committed to dev version.

Thanks for the update.

closed 2018-03-18 by repetier