repetier/RepetierFirmware #757: Support for polar printers?

vanbins, 2018-03-01:
Hi! I'm working on making a polar printer, and right now the gcode for it is being made manually. The machine is already using Repetier and it's working fine (still needs a lot of tweaking, but it prints), but it would be great if "regular" gcode could be used, made by a slicer. Converting Cartesian  coordinates into polar ones isn't the thing i'm struggling with that much, but converting the speed and extruder feedrate is what is complicated. The printer has some unusual "features" right now. In example, One unit on the Y doesn't stand for 1 degree of movement as most polar systems, but rather for 1 rotation of the turntable.
Could polar kinematics be added to Repetier firmware?

repetier, 2018-03-01:
If you can, you should start directly with dev2 branch of our firmware. It already works in due based system, just not with display and z probe is currently not working, but we work on that.

Here we have a brand new system of handling different printer types and you could just write your own printer type for polar printers. All printer dependent actions are now bundled in one class, so it should be quite easy to do so. All moves are computed in cartesian space and all you have do do is to offer transformation cartesian position->motor position. Also homing often needs tweaking with such special things, so homing is also a printer dependent function.

Only thing that might be a problem is near rotation axis - here rotation speed could go to infinite and that would not work. Not sure how you solved that problem.

vanbins, 2018-03-01:
Thanks! I'll take a look at it