repetier/RepetierFirmware #759: Heating manager with deadtime ist not working correctly

MacFlieger, 2018-03-05:
With Repetier-Firmware 0.92 I have used dead time control as heat manager for Hotend and Bed and it worked well, as you can see in the following image:
[image]

But with version 1.0.1 and the same settings it is not working correctly.
- The temperature of the Honend is more unstable.
- The temperature of the bed climbs up to 20° over the target temperature and then cools down to the correct one.
[image]

The settings of 0.92.9:
[image]
The settings of 1.0.1:
[image]

repetier, 2018-03-07:
We had changed the temperature measurement a bit so that the mostly used PID works much better. But now the resolution for dead time has decreased from the same change so that dead time is in deed not as good as before.

In V2 we are more flexible and I hope I can get back to the old behaviour or even better by having also a cooling dead time.

MacFlieger, 2018-03-07:
OK, thanks.

Because of the very bad behavior of the bed temperature, I have also tried PID with and without Autotune.
I have tried all 5 methods of autotune, but the results were very bad, more bad than the above dead time.
So I manually optimized the PID and I got parameters, which are  OK, but clearly not so good as the old dead time behavior.

repetier, 2018-03-07:
Depends on definition of better behaviour I'd say. PID will always have it's inswing problem, so it will overshoot or undershoot and take time to stabilize. That is clearly the nice part on dead time. On the other side PID will convert to a more constant curve in general while dead time MUST swing, also it tries of course to do only very slightly. Here PID can converge into a more narrow band.

MacFlieger, 2018-03-07:
Yes, I agree.
With my manual optimized PIDs, there is a little bit inswing (around 1°), but after that the temperature is more stable. It is usable.
The results of autotune is completely unusable with great inswing and/or long time very wrong temperatures. The I-component is always much higher than the manual optimized one.