repetier/RepetierFirmware #762: V2 want to test

boelle, 2018-03-11:
I'm currently using the dev branch

https://www.repetier.com/firmware/dev/index.php

but i'm very interested in the new disortion correction in V2 that you have told us about

is there anything special i should do to test it? does it share the same configuration file or is it better just to wait until its released the "prober" way?

repetier, 2018-03-11:
Apart from z probing/distortion currently not working it has a completely new config system, because it is much more modular and flexible. Also it can print on some systems many functions need to be adapted still and documentation is also not that much. So at the moment you should better wait at least until z probing/distorion is working..

boelle, 2018-03-11:
oki... i assume we can use this "issue" as a kind of signup list for early testers if i edit the subject? i will also edit down my first post to only have the very essientals data

boelle, 2018-03-11:
btw i think i have found an error in dual z driver with dual Z endstop

both endstops work when i activate them with a finger, both Z steppers move when i use repetier host

but with homing stepper 1 does not stop when endstop is activated. i posted on forum too (sorry for double posting)

repetier, 2018-03-11:
Where I really add infos about V2 is the readme in V2. I try to hold it in sync with the changes done and outstanding. So if you watch the project you should get these.

closed 2019-03-19 by boelle