repetier/RepetierFirmware #764: Z-Probe problem with X-Y-position in Z-Probe-End-Skript

viper-666, 2018-03-21:
Hello,
I have the following issue:
My Z-Probe has to be activate/deactivate by moving to a certain X/Y position.
First I homed Xmin. , Ymin. and Zmax. then I go to X100 and Y100 because my Z-Probe has an offset to the nozzel and if I want to use the Z-Probe on X0 and Y0 it gives an error that the Z-Probe can't reach the position. That is right if you go from X0 and Y0 but when the Z-Probe-Start-Skript first be done I can bring the nozzel in the right position. But this is an other issue...

The Issue I want to open here ist when I now give the G30 command the G-Code possition Y292 and X0 ind the Z-Probe-Start-Skript are approached correctly.
After that the nozzel moves back to the startpoint and then to the Z-Probe offset position and the bed moves up and after probing down.
But now the possitions in the Z-Probe-End-Skript will not be reached. I want to go to Y292 and X242 But it looks like he's subtracting the difference from the Z-Probe offset in X and Y.
When the G30 command is finished the X and Y positions are correct again.

repetier, 2018-03-21:
No so sure I understand your problem, since you say after G30 it is correct. During probing you always have the z probe offset active but position stored will not change. Of course you are not allowed to home in start script, that would cause errors here. But you can move to activate position.

Also note that Px in G30 can tell if start/end script is run or not, in case that this would help.

viper-666, 2018-03-21:
Start and End Script are running. In the Start Skript the positions Y292 and X0 where approachted correcly and my Z-Probe will activate.
But after Probing the G-Code-Positions Y292 and Y242 were not reached so my Probe were not deactivate.
It seems he moves to Y292-42 (42 is the Y offset from the probe to the nozzel) and X242-12 the X offset of the nozzle.
When G30 is ready the X and Y positions are correct again. So if I give the command G1 Y292 and G1 X242 the probe were deactivate.
I think it has to do with the offset move it do to set the probe on the point were the nozzle was before he probes the bed high.

repetier, 2018-03-21:
Ok, now I understand. Yes, I think you are right. When running the end script the offset is active so you would need Y334 instead. If you have problems because it says it is unreachable use G1 S1 Y334\nG1 S0 so it disables boundary check for disabling so you can reach the spot.

viper-666, 2018-03-21:
Ok, the S1 command do the trick. With this workaround I can go further.