repetier/RepetierFirmware #765: Enabled TMC2130 SPI / RAMPS and hangs on startup (1.0.1dev)

badvision, 2018-03-23:
I'm trying to use TMC2130 SPI (wired to the SPI pins on Aux 2) and I have a Melzi screen wired up to Aux 1 using software SPI (#define U8GLIB_ST7920)

most of the time I can't get the arduino to boot.  A couple of times I get some basic info about jerk settings being too high but then it's just frozen.  It never gets as far as the splash screen part, just a blank screen.  If I flash over a working firmware, it never even clears the screen, so that leads me to believe it hangs before initalizing any of the UI.

Only other things of note are that I enabled sensorless homing and for X and Y, I picked the pins for X+ Endstop and Y+ Endstop, respectively.  I don't have those pins configured for anything else currently.

With no display and, generally most of the time, no messages on the com port, I'm really at a loss at how to debug this further.

Thoughts?   Suggestions?

Thanks!

repetier, 2018-03-23:
Is this a TMC issue? I mean does it start if you say you have normal drivers?

badvision, 2018-03-23:
If I enable TMC it hangs on start.  Literally just commenting out the TMC2130 feature flag in configuration.h at this point and I see it start again, granted without TMC support.   Even if I had a wiring problem (not likely) I wouldn't expect the whole program to freeze up like that.

badvision, 2018-03-23:
For simplicity I'm running this on a bare board with nothing plugging in except the melzi screen, still hangs.

repetier, 2018-03-24:
Since I did not write the TMC addition I can not really say if it can hang, but I guess it could if the drivers wait for a response that never comes for example because communication is not working. I send this link to the author, maybe he can say more.

badvision, 2018-03-24:
It would be unlikely for the software spi of the screen to cause problems, is that a fair assumption?

repetier, 2018-03-24:
Yes, especially if they do not share a pin there should be no interference.

alexsomesan, 2018-03-24:
@badvision I'm the guy who added the TMC support. The symptoms you're describing are usually caused by the SPI communication not working, most likely becuase of wrong connections. The same happened to me a few times when wiring new boards.

Can you explain exactly how you wired you drivers? Even better, can you post some pictures of the board and the wiring?

We also have a forum thread where we're discussing issues specific to the Trinamic drivers. You can post pictures there if it's easier: https://forum.repetier.com/discussion/1897/support-for-new-tmc2130-sss-silentstepstick-via-spi

NojbyMichal, 2018-03-24:
Hi, I am going to add some facts to the problem about TMC and SPI. I am having same problem with booting repetier. The problem is about the SD card and TMC are sharing SPI. When I start the printer with sd card plugged in, it wont boot up. The LCD 20x4 is blank, After that I remove the SD card and it boots normally. So the problem is when the SD card is plugged in when it try to boot.

alexsomesan, 2018-03-24:
The SPI bus itself is designed to work with multiple devices connected. Only one device can be active on the bus at any one time and this is signaled by the CS lines going to each device. CS lines cannot be shared among devices, but MISO, MOSI and CLK are shared.

Is your screen using the same MISO, MOSI and CLK lines as the stepper drivers?
Which pins did you choose for the CS lines of each of the steppers?
Which CS line is the screen using? How is the screen controller connected to RAMPS?

NojbyMichal, 2018-03-25:
alexsomesan: I am electrician and programmer, so this basics are well known for me. MISO, MOSI, CLK are shared for TMC and LCD, CS has every unit his own. CS: X-driver 42, Y-driver 44, Z-driver 66, SD card as original for RAMPS. LCD is connected via PCB reduction for RAMPS, for two flat cable.

badvision, 2018-04-01:
Sorry, the SPI bit was a red herring.  I updated the code to use Software SPI and got the same result.

Remove/comment out the following line of code that created the infinite loop:
  void Printer::configTMC2130(TMC2130Stepper* tmc_driver, bool tmc_stealthchop, int8_t tmc_sgt,
      uint8_t tmc_pwm_ampl, uint8_t tmc_pwm_grad, bool tmc_pwm_autoscale, uint8_t tmc_pwm_freq) {
        //while(!tmc_driver->stst());                     // Wait for motor stand-still
        tmc_driver->begin();                            // Initiate pins and registeries

This might be my own lack of knowledge, but how the hell can we get status from the driver before we init the chip?   Seems the best course of action is to not wait forever in this case.  Some kind of boundary check or iteration limit would at least be prudent here!

badvision, 2018-04-01:
Also the TMC2130 driver constructor has an alternate that allows software SPI by just providing the three extra pins as parameters.  Seems like a simple thing to support for little effort.

badvision, 2018-04-01:
Here's how I've wired it up on my RAMPS 1.5 board:

Driver | Ramps | Pin |  
-- | -- | -- | --
SDO (X,Y) | MISO | D50 |  
SDI (X,Y) | MOSI | D51 |  
SCK (X,Y) | SCK | D52 |  
CS (X) | X+ Stop | D2 |  
CS (Y) | Y+ Stop | D15 |

badvision, 2018-04-01:
As for other question about screen control - I have that all run to the AUX-2 connector since I was too lazy/cheap to buy a new LCD screen when I ditched my Melzi board.  If you swap the +5/GND wires you can run a Melzi off of Aux2 on RAMPS.  You have to weak the UI config a little bit and configure the software SPI pins, but not too bad.

NojbyMichal, 2018-04-14:
I solved the problem about printer not booting up because of conflict on SPI between SD card and TMC2130. @repetier could you make a update in  file:  Printer.cpp in function: void Printer::configTMC2130(TMC2130Stepper* tmc_driver, bool tmc_stealthchop, int8_t tmc_sgt,
      uint8_t tmc_pwm_ampl, uint8_t tmc_pwm_grad, bool tmc_pwm_autoscale, uint8_t tmc_pwm_freq) {
make comment to  while(!tmc_driver->stst());  (this line needs to be commented  -> // while(!tmc_driver->stst()); ) That works for me with this issue.

badvision, 2018-04-16:
Yeah -- I don't think it's a SPI conflict.  I think that tmc_driver->stst() cannot be called before the driver is initialized.  It is probably just the driver starting in an unknown state.  But I've gone through many cold-starts with that line commented out and no more problems here.

NojbyMichal, 2018-04-16:
@badvision The problem is only in one situation, and that is: Powering on the printer with inserted SD card,. I don't know if you (@badvision) catch that information. When I start the printer without plugged SD card, everything works. But with plugged SD card the printer is not able to start, LCD has only backlight and via USB/UART I dont see any log

alexsomesan, 2018-04-16:
Hi everyone,
@badvision is right! 
Indeed `begin()` has to be called first. I ran into this issue myself recently while working on the TMC initialisation for Repetier V2. 
I ported over the initialisation from V2, which does the right sequence and also uses a timeout instead of the dumb`while();`

You can find it here: https://github.com/alexsomesan/Repetier-Firmware/tree/improve-trinamic-init
Give those changes a try and let me know if they work correctly. I am on a plane now and cannot test it.

@NojbyMichal Checking for standstill is still needed (the `stst()` call) because some parameters can only be set while the drivers are at standstill - please don't delete it. Normally it returns very fast as it's just a register comparison. Try out the changes in the branch I linked. I will make a PR to the main repository once you guys confirm it's working for you.

Thank for your patience! I'm travelling a lot lately so I may react a bit slower here.

Cheers,
Alex

badvision, 2018-04-16:
Sweet!   Out of curiosity, I see how to configure the CS pins but I don't see any ability to configure the extra pin (diagnostic pin) to support features like the stallguard error flag to detect missed steps.  Am I missing something?   I know that it was supposed in Marlin but I'd rather not switch to Marlin just to have that feature.

NojbyMichal, 2018-04-17:
@badvision  I think that it is done reading SPI register for stallguard, Marlin has ported this information from that register to one from DIAG0 or DIAG1 pins, it can be done in repetier too, but my mind, there is no support how to react on this digital inputs. For others developers, If I am wrong, let me know.

NojbyMichal, 2018-04-17:
@alexsomesan  Alex, how about SENSORLESS_HOMIG for delta ?

alexsomesan, 2018-04-17:
Detecting missed steps is not implemented yet. It requires more extensive changes to he movement routines.
Hardware wise it is not complicated: the TMCs have a register which holds the actual number of missed steps detected. This can be read via SPI. But continually reading this while also performing motion computations is again impractical, so there is an option to configure the other Diag0 pin (or even Diag1) to send a signal that skipped steps were detected. It is then more practical to react to this signal (via interupt) and read the skipped steps value.

This is all simple: starts the complicated part. What to do with this value and how (if) it can be integrated in the motion planner. 
Potentially it could be used to compensate by adding more steps to the current movement but it’s tricky because it will upset the acceleration profile which was already computed. The simpler thing to do is rehome and resume from the curent actual coordinates (like Prusa does).

I’m planinng to experiment with this after all the basic features are settled in V2. But before that I would like to focus on more immediately beneficial things like an automatic tunning command for newly installed boards (similar to PID tuning) so that users don’t have to guesstimate so many Trinamic parameters as they do today.

cubical23, 2018-07-05:
I'm using an arduino due & RADDS 1.5 board, I also have boot problems. But it only happens, when power is provided via USB only. When powering via RADDS / big power supply, everything will work as expected.

When disabling TMC2130 support in firmware, there are no boot problems at all.

gregbiv, 2018-10-04:
I can upvote on this, I have Arduino Ramps 1.4 and TMC2130 drivers with Repetier firmware 1.0.3 installed. When I plug in only by USB I see an empty screen, but when I turn on the power supply - the screen with booting Repetier appears immediately

gregbiv, 2018-10-04:
@qx1147 it works with the fix made by @alexsomesan in https://github.com/alexsomesan/Repetier-Firmware/tree/improve-trinamic-init

closed 2022-01-05 by badvision