repetier/RepetierFirmware #767: Compile fails with configuration for Wanhao i3 / Monoprice Maker Select

cmot17, 2018-03-28:
The error I get is this:
`'DIO49_DDR' was not declared in this scope`

Set the board to Melzi and select the RepRapDiscount LCD. 
It seems to be some kind of issue with SDCARDDETECT, but I really can't figure out what's going on.
Full error message:
`In file included from sketch/HAL.h:89:0,
                 from sketch/Repetier.h:597,
                 from sketch/Printer.cpp:19:
sketch/Printer.cpp: In static member function 'static void Printer::setup()':
fastio.h:34: error: 'DIO49_DDR' was not declared in this scope
 #define  _SET_INPUT(IO)  do {DIO ##  IO ## _DDR &= ~MASK(DIO ## IO ## _PIN); } while (0)
                              ^
sketch/fastio.h:54:26: note: in expansion of macro '_SET_INPUT'
 #define  SET_INPUT(IO)   _SET_INPUT(IO)
                          ^
sketch/Printer.cpp:904:5: note: in expansion of macro 'SET_INPUT'
     SET_INPUT(SDCARDDETECT);
     ^
In file included from sketch/HAL.h:89:0,
                 from sketch/Repetier.h:597,
                 from sketch/Printer.cpp:19:
fastio.h:34: error: 'DIO49_PIN' was not declared in this scope
 #define  _SET_INPUT(IO)  do {DIO ##  IO ## _DDR &= ~MASK(DIO ## IO ## _PIN); } while (0)
                                                          ^
sketch/fastio.h:17:30: note: in definition of macro 'MASK'
  #define  MASK(PIN)    (1 << PIN)
                              ^
sketch/fastio.h:54:26: note: in expansion of macro '_SET_INPUT'
 #define  SET_INPUT(IO)   _SET_INPUT(IO)
                          ^
sketch/Printer.cpp:904:5: note: in expansion of macro 'SET_INPUT'
     SET_INPUT(SDCARDDETECT);
     ^
fastio.h:29: error: 'DIO49_WPORT' was not declared in this scope
 #define  _WRITE(IO, v)   do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)
                                       ^
sketch/fastio.h:49:37: note: in expansion of macro '_WRITE'
 #define     PULLUP(IO,v)            _WRITE(IO, v)
                                     ^
sketch/Printer.cpp:905:5: note: in expansion of macro 'PULLUP'
     PULLUP(SDCARDDETECT, HIGH);
     ^
fastio.h:29: error: 'DIO49_PIN' was not declared in this scope
 #define  _WRITE(IO, v)   do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)
                                                                    ^
sketch/fastio.h:17:30: note: in definition of macro 'MASK'
  #define  MASK(PIN)    (1 << PIN)
                              ^
sketch/fastio.h:49:37: note: in expansion of macro '_WRITE'
 #define     PULLUP(IO,v)            _WRITE(IO, v)
                                     ^
sketch/Printer.cpp:905:5: note: in expansion of macro 'PULLUP'
     PULLUP(SDCARDDETECT, HIGH);
     ^
fastio.h:29: error: 'DIO49_WPORT' was not declared in this scope
 #define  _WRITE(IO, v)   do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)
                                                                                                ^
sketch/fastio.h:49:37: note: in expansion of macro '_WRITE'
 #define     PULLUP(IO,v)            _WRITE(IO, v)
                                     ^
sketch/Printer.cpp:905:5: note: in expansion of macro 'PULLUP'
     PULLUP(SDCARDDETECT, HIGH);
     ^
fastio.h:29: error: 'DIO49_PIN' was not declared in this scope
 #define  _WRITE(IO, v)   do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)
                                                                                                                              ^
sketch/fastio.h:17:30: note: in definition of macro 'MASK'
  #define  MASK(PIN)    (1 << PIN)
                              ^
sketch/fastio.h:49:37: note: in expansion of macro '_WRITE'
 #define     PULLUP(IO,v)            _WRITE(IO, v)
                                     ^
sketch/Printer.cpp:905:5: note: in expansion of macro 'PULLUP'
     PULLUP(SDCARDDETECT, HIGH);
     ^
exit status 1
'DIO49_DDR' was not declared in this scope
`
[Repetier-Firmware-2018-03-28.zip](https://github.com/repetier/Repetier-Firmware/files/1856025/Repetier-Firmware-2018-03-28.zip)

repetier, 2018-03-28:
Melzi boards don't have pin 49. I guess you selected a display that has no settings for this board and it did fall back to an other board where sd detect uses pin 49.

jamesarm97, 2018-03-28:
Correct, SD Detect is not usable on the Melzi and is why you have to unmount and remount the SD card when you make changes. In my Pins.h for board 63 (you may be able to add to your configuration.h):

#define ORIG_SD_CARDDETECT -1

cmot17, 2018-03-28:
That is defined in my firmware already in the file pins.h under the section for the Melzi board. Is there another place where this could be redefined?

cmot17, 2018-03-28:
I attached my firmware in the original post if that might help.

repetier, 2018-03-28:
Yes, as I said if you select a display not configured in DisplayList.h for melzi you will end up with wrong pin settings - and many displays include a own sd card hence have a own sd detect pin as in this case.

jamesarm97, 2018-03-28:
Does the Configuration.h file override most stuff? Can he undef then define the detect pin as -1 in the Configuration.h so the standard .h files are not modified?

repetier, 2018-03-28:
Not in this case. DisplayList.h overrides config - he has to us ethe right display or write display config on his own if it is not defined for melzi to fit his pin numbers.

cmot17, 2018-03-28:
I'm sorry if this is obvious, but I'm honestly at a loss right now. How can I write a new display config? I can't even find where the SDCARDDETECT pin is defined for my display - or where the fall back to default code is. I've looked all through displaylist.h. Thanks for your help.

jamesarm97, 2018-03-28:
The Melzi doesn't even have the two headers for the RepRap LCD. How are you wiring this up? The Melzi has an onboard SD card. You could wire up the Reprap LCD only on one header and not use the LCD SD. Are you trying to connect the original display or your own, because the original looks like the RepRap Discount LCD but without the SD card slot on it. Probably best to move this over to some other chat area since this is just for issues / changes to the code.

cmot17, 2018-03-28:
This is a Wanhao i3 / Monoprice Maker Select v2.

Sorry if this is the wrong place to ask, where do you think I could go instead?

repetier, 2018-03-29:
I think having a sold standard model google for the configuration might help here. We do not have a config for this, but maybe someone else has published it. Repetier Forum might also have a visitor having it, but google will be faster.

jamesarm97, 2018-03-29:
https://groups.google.com/forum/#!topic/wanhao-printer-3d/ZhPeAmiLRak%5B126-150%5D
https://3dprinterwiki.info/wanhao/part-specifications/#Firmware

cmot17, 2018-03-29:
I googled around a bit, but all I find is older versions, which seem to not even have the DisplayList.h file. I have attached a working configuration (but it's an older version). I've been comparing the two and I can't find the issue.
[Repetier-Firmware-2016-07-19.zip](https://github.com/repetier/Repetier-Firmware/files/1860482/Repetier-Firmware-2016-07-19.zip)

badvision, 2018-04-01:
Since you're in there, just ditch the Melzi board. :)  I converted my printer to run off a RAMPS and couldn't be happier.  Got the Melzi display working with it too.  Check my fork for more information.