repetier/RepetierFirmware #779: Nextion LCD display support?

FlameFlameFlame, 2018-04-19:
Hello!
I use RuRamps4D with Reperier-Firmware configured via website. I'm plugging in Nextion into dedicated UART port but there is no communication between Nextion display and printer. Am I missing something in configuration file? I enabled bluetooth communication on Serial 1 and set BAUD rate to 115200, such as Nextion.

repetier, 2018-04-19:
Do you have a program running on nexion that does talk to firmware?

As a first test I would try to connect each port to repetier-host to see if they work. If they do you know it is the nexion software making the problem.

jamesarm97, 2018-04-19:
Has anyone done a Nextion add-in for Repetier? I don't see one in the UI list of displays. The Nextion runs its own firmware and the repetier board will need to know how to talk to it with specific commands. I know Marlin has some support on specific branches for the Wanhao I3 Plus printer. It is also not generic, someone would have to design an interface specifically for 3d printing with the matching repetier firmware changes.

FlameFlameFlame, 2018-04-20:
@repetier Yes, I do have a programm on Nextion. It uses Marlin-flavored G-codes, BTW, but as I could understand Repetier Firmware uses it too.

FlameFlameFlame, 2018-04-20:
@jamesarm97 There is no need in some special program on printer. Nextion works as a computer in this case, it just sends G-codes to printer using serial communication.

jamesarm97, 2018-04-20:
Ok, that makes sense. All the Nextion displays I have had firmware that sent packet data that had to be parsed on the repetier / marlin side (Wanhao I3 Plus and the ADV++ firmware). I would love to replace their code with something easier to work with. What is the firmware you are using on the Nextion display?

FlameFlameFlame, 2018-04-20:
For a Nextion I use this firmware: https://github.com/apballard/Marlin

FlameFlameFlame, 2018-04-26:
@repetier is there any way to define several serial ports in Repetier firmware? As far as I understand there is one serial port to connect with PC, is it possible to enable another one to connect with Nextion?

FlameFlameFlame, 2018-04-26:
> Yes it is no problem to have 2 serial ports in parallel and also both
> being used. But the user needs to know what he does e.g. not printing on
> both ports as that would cause unwanted moves.
>
> In config tool this is names as bluetooth serial, but in fact it is just a
> second serial that might be connected to a serial bluetooth converter. This
> only works on V1 as this is the first to separate line numbers per
> connection.

Thank you for reply :)
Which pins bluetooth serial uses? 18/19? And how do I define baudrate? Is
it similar with default serial?

repetier, 2018-04-26:
It uses the pins of the selected serial port. These are hard coded. Baudrate is set by you as well see config description for it. In the due pinout you can see which serial port uses which pins.

/*
We can connect BlueTooth to serial converter module directly to boards based on AtMega2560 or AtMega1280 and some boards based on AtMega2561, AtMega1281 or AtMega1284p
- On Melzi boards connect BT to TX1 and RX1 pins, then set BLUETOOTH_SERIAL to 1
- On RUMBA boards connect BT to pin 11 and 12 of X3 connector, then set BLUETOOTH_SERIAL to 3
- On RAMBO boards connect BT to pins 5,6 or 7,8 or 9,10 on Serial connector, then accordingly set BLUETOOTH_SERIAL to 1,2 or 3
- On RAMPS we must remap Y_ENDSTOPS pins or Z_ENDSTOPZ pins or LCD_ENABLE and LCD_RS pins to another pins, and connect BT to:
  a) signals of Y_MIN, Y_MAX, then set BLUETOOTH_SERIAL to 3 (RX from BT to Y_MIN, TX from BT to Y_MAX)
  b) signals of Z_MIN, Z_MAX, then set BLUETOOTH_SERIAL to 1 (RX from BT to Z_MIN, TX from BT to Z_MAX)
  c) pin 17 and 18 of AUX4 connector, then set BLUETOOTH_SERIAL to 2 (RX from BT to AUX4 p18, TX from BT to AUX4 p17)
  Comment out or set the BLUETOOTH_SERIAL to 0 or -1 to disable this feature.
*/
#define BLUETOOTH_SERIAL   1                      // Port number (1..3) - For RUMBA use 3
#define BLUETOOTH_BAUD     115200                 // communication speed

repetier, 2018-04-26:
You can not change the serial pins - they are processor specific. But depending on the cpu type there might be more then 2 serial ports.

FlameFlameFlame, 2018-05-03:
Could you please tell in which file can I configure CPU specific pins?

vinmicgra, 2018-05-03:
As repetier said use the serial pins according to your boards pinout diagram. You can easily define second serial in the config tool.
This is an issue-tracker for code-related stuff and not a support forum. Please stop wasting time of developers with beginner questions. There are many helpful forums and social-media groups.

FlameFlameFlame, 2018-05-03:
Okay, it was Nextion fault. Tried MKS-TFT and it worked flawlessly.

closed 2018-05-03 by FlameFlameFlame