repetier/RepetierFirmware #781: Advance 1.5?

cfelicio, 2018-04-24:
Have you seen this new feature on Marlin?

https://github.com/MarlinFirmware/Marlin/pull/9700

Wondering if it's something Repetier would consider porting over :)

repetier, 2018-04-25:
In V2 we have already reworked linear advance to be more smooth using our new motion control system. That they have changed unit of K has no direct effect on quality. Need to check what the difference is, but I guess it is the cross section and a scaling factor. Will have to think about what makes more sense to get comparable results.

Last thing is no retraction during linear advance. That is a fine thing. But that can result in very slow accelerations. In V2 we have instead reduced retraction to no extruder move as limit. What is better is something we need to test in future and might also be biased by user preference. Linear advance makes most sense when printing fast while limiting acceleration for this constrain would just prevent fast moves, so you have a contradiction as only one condition can be fulfilled here. This will be especially apparent for bowden extruders requiring much more speed. But maybe I interpret this wrong

        const uint32_t max_accel_steps_per_s2 = max_jerk[E_AXIS] / (extruder_advance_K * block->e_D_ratio) * steps_per_mm;
+          #if ENABLED(LA_DEBUG)
+            if (accel > max_accel_steps_per_s2)
+              SERIAL_ECHOLNPGM("Acceleration limited.");
+          #endif
+          NOMORE(accel, max_accel_steps_per_s2);

not so sure how this limit is supposed to work. 

In any case I hope that V2 will improve linear advance another step.

cfelicio, 2018-04-25:
Awesome, thanks for looking into this! Looking forward to Repetier V2! :-)