repetier/RepetierFirmware #783: PID autotuning doesn't turn on the ATX PSU

jonas-schievink, 2018-04-28:
Sending `N5 M303 S200 C8*60` to the printer (via OctoPrint) results in the following response:

```
ok 5
Info:PID Autotune start
```

followed by `busy:heating` messages. However, the ATX PSU wasn't turned on, resulting in no heating at all. Additionally, the firmware didn't seem to care about that fact. I would expect the thermal runaway protection to kick in here.

This is on Repetier 1.0.1, with [this Configuration.h](https://github.com/repetier/Repetier-Firmware/files/1957922/Configuration.h.txt).

repetier, 2018-04-29:
Please use repetier-firmware syntax

- M303 P<extruder/bed> S<printTemerature> X0 R<Repetitions> C<method>- Auto detect pid values. Use P<NUM_EXTRUDER> for heated bed. X0 saves result in EEPROM. R is number of cycles.
				method 0 = classic, 1 = some overshoot, 2 = no overshoot, 3 = pessen, 4 = Tyreus-Lyben

Especially use P to tell what you want to calibrate and use valid C values.

jonas-schievink, 2018-04-29:
Oh, okay, I took the command from http://reprap.org/wiki/G-code#M303:_Run_PID_tuning

It's odd that it worked correctly after turning on the PSU from the menu, I'd expect the firmware to complain if the syntax is invalid.

(I only now noticed that all Repetier G-Codes are documented in [Repetier.ino](https://github.com/repetier/Repetier-Firmware/blob/178d5b73668d90d59f8910a2a1099b0b18d99586/src/ArduinoAVR/Repetier/Repetier.ino#L144-L145))

```
- M303 P<extruder/bed> S<printTemerature> X0 R<Repetitions> C<method>- Auto detect pid values. Use P<NUM_EXTRUDER> for heated bed. X0 saves result in EEPROM. R is number of cycles.
				method 0 = classic, 1 = some overshoot, 2 = no overshoot, 3 = pessen, 4 = Tyreus-Lyben
```

repetier, 2018-04-30:
I complained because it was wrong, so it could be the reason. Of course not having power is a even worse reason and fully explains why it did not work. Closing issue.

closed 2018-04-30 by repetier

repetier, 2018-04-30:
Sorry, I did not understand what your intention was and having no ATX I can never test it.

void TemperatureController::autotunePID(float temp, uint8_t controllerId, int maxCycles, bool storeValues, int method) {
	ENSURE_POWER

will be added with next update to ensure it powers on in that case. Thank you.