repetier/RepetierFirmware #784: No ".." (go back to parent directory) when browsing SD card from LCD

themelle, 2018-05-02:
Previously (0.92.x) there alsways has been a way to go back to the parent directory when in SD card browsing mode in LCD UI. 
As of 1.0.x this possibility is - to my knowledge - gone. I can browser down the directory tree, but the top entry ".." to go back is always missing. Only way to go "back" is to eject and re-insert SD card, which is not the way its meant to be played ... :-)

Or am I missing something obvious?

Nibbels, 2018-05-10:
Hello @themelle ,

First I thought that now the visited folder is remembered and not cleared up correctly.
But at least I have two "gobacks". One is by button and one by the most top menu point. So I liked to navigate with the most top menu point and drop out when I am in folderlevel 0.

https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoAVR/Repetier/ui.cpp#L2501

I went to the place where the top link points to and changed the "goback". Checking the folderlevel and goDir("..") does the magic.

```
        if((menuPos[menuLevel] == 0 && folderLevel == 0) /* Selected back instead of file */
            || !sd.sdactive )                            /* No SD -> drop out of the folders listing */ 
        {
            executeAction(UI_ACTION_BACK);
            return 0;
        }
        if(menuPos[menuLevel] == 0 && folderLevel > 0){
            goDir("..");
            menuTop[menuLevel]=0;
            menuPos[menuLevel]=1;
            refreshPage();
            return 0;
        }
```

See https://github.com/repetier/Repetier-Firmware/issues/791 for some more tweak.

But I cannot garantee that the code atop works with the active 1.0.1 repetier version, because what I do is patching a quite old split branch from conrad electronic, not the active development branch.
Maybe repetier himself knows what to do best.

Greetings from Stuttgart

rossiniscarface, 2018-05-28:
This is still a problem in v1.0.2