repetier/RepetierFirmware #787: Printing from SD card sometimes not possible

themelle, 2018-05-07:
When trying to print files from SD card sometimes after pressing the knob to select a file to start printing there is no reaction at all. The browse dialog stays open and one can turn the know to scroll to a different file.
After shortening the not-printable file's name printing the same file works as expeced, so I assume there is some file name length restriction kicking in and preventing starting prints for files with names above a certain length.
Will do some more testing to find out the exact limit for file name lengths.

themelle, 2018-05-07:
Thanks for the quick response.
Printing is started exclusively through LCD + encoder knob, no host involved. Though a limitation seems to be in place.

repetier, 2018-05-07:
Looking into the code

```
        switch(shortAction) {
        case UI_ACTION_SD_PRINT:
            if (sd.selectFile(filename, false)) {
                sd.startPrint();
                BEEP_LONG;
                menuLevel = 0;
            }
            break;
```
I see it is also using filename here, so guess the same 26 char limit defined in repetier.h (2 13 char blocks) is in effect here.

themelle, 2018-05-08:
Hmm, 
did this code change at some point between 0.92.x and 1.0.x? 
I can not remember having this problem with 0.92.x ...

Nibbels, 2018-05-10:
I too tested a file with a quite long name:

0:15:28.295: SD listing: 4 / 2
0:15:45.039: SD listing: 1 / 2
0:15:46.397: file.open failed
0:15:53.129: SD listing: 2 / 2
0:15:53.528: SD listing: 3 / 2
0:15:54.216: SD listing: 4 / 2
0:15:54.888: file.open failed
0:15:59.652: SD listing: 3 / 2
0:16:00.004: SD listing: 2 / 2
0:16:01.474: SD listing: 1 / 2 (2)
0:16:02.784: File opened:FrontWand.gcode Size:8216407
0:16:02.784: File selected
0:16:12.522: SD print stopped.

SDCard::selectFile sais file.open failed.
Else we should get some 8.3 pendant for starting prints from SD (by menu) or increase this limit. I guess..

Greetings

Nibbels, 2018-05-11:
Ok, 

what I did: I merged some functions in quick and dirty style to get a 
`bool selectFileByPos(uint16_t filePos, bool silent=false);`
as alternative to the old 
`bool selectFile(const char *filename,bool silent=false);`
https://github.com/Nibbels/Repetier-Firmware/blob/ac4607382555dbe442633d2b7261df925be594c0/Repetier/SDCard.cpp#L388

This might still be buggy or way to complex but as a first shot it seems to work.
Disclaimer: I understood a lot, but not everything when it comes to static SdFat fat; SdFile file; and some "use and waste" objects like FatFile file; (and when to use which)

https://github.com/Nibbels/Repetier-Firmware/blob/ac4607382555dbe442633d2b7261df925be594c0/Repetier/Repetier.h#L206
https://github.com/Nibbels/Repetier-Firmware/blob/ac4607382555dbe442633d2b7261df925be594c0/Repetier/ui.cpp#L2747
I did not make the delete function work for me yet. Instead I just killed it. So I cannot delete files using my printers menu.

(And I did not understand why automount prechooses "Print File" for me.
https://github.com/Nibbels/Repetier-Firmware/blob/ac4607382555dbe442633d2b7261df925be594c0/Repetier/ui.cpp#L2732)

Well, but this is it for now.

Greetings