repetier/RepetierFirmware #788: BED_LEVELING_METHOD == 1 // linear regression : Probe is not probing the wrong area

sboettger, 2018-05-07:
As I kept failing using the linear regression feature, I took a look into the code. As far I understand, the linear regression feature is not working correct. I proved this by writing the code in php and creating images with the Probe-coordinates and the 3 spanning coordinates:

https://gist.github.com/phpfiddle/d3b482a343eb16de23fbaa03ddd1cdee

Assumption: beside delta printers, every printer has a rectangular bed, and needs a flat square probe area inside his dimensions. Therefore we can make the method much more easier:

I will attempt to change the BED_LEVELING_METHOD now, because I think its far to complicated. You don't need a square defined by these 3 points, but by a reference coordinate (1 point) and a corresponding height and width of the flat square
[image]
 should fit all requirements.

sboettger, 2018-05-08:
fixed by #789

rossiniscarface, 2018-05-28:
Hi,

i think you have your 3 coordinates in your php-example wrong. 
P1 should be x1/y1 30/30
P2 should be x2/y2 170/30
P3 should be x3/y3 30/170

Then the rectangle should be right.

sboettger, 2018-05-29:
Hmm, yes you're right. But don't you think my method is still easier?

rossiniscarface, 2018-05-30:
I think not, because you give some eeprom-variables another semantic. 
Sometimes they are points, and sometimes their parts are height/width.

repetier, 2018-05-30:
Also the bending corrections are at the coordinates. So while you say you only use 4 of 6 values you would still need to set all 6 correctly. And if they are set correctly it makes no difference. The problem is mainly that you can define what the 3 points are used for and in that ver special case you can omit 2 values if you know what you are doing and compute the 2 from the rest. But what if P1 is at the right top, then you have to watch sign and go to left/front instead. 

For V2 I'm thinking about complete remove the  area and instead define just where the bed would be so it probes automatically the complete bed if reachable or the part reachable. And only for other leveling methods you would enter points then. But that requires the new config system of V2 to implement.