repetier/RepetierFirmware #793: bl-touch probe missing pulse signal

cor-mt, 2018-05-31:
On my printer I got the bl-touch configured, the servo commands are working as they should. The same hardware is working with marlin. Now when I try "G30" it moves down till touching the bed, giving it's pulse witch repetier misses, pulling back the probe and then when it drop's it again the printer is still going down so the probe goes into error mode. In error mode the signal stays high so then repetier does pick it up and backtrack but the probe now stay in error mode and the printer fail with "Error:z-probe did not untrigger on repetitive measurement - maybe you need to increase distance!"

From the forums I sea Marlin had the same problem witch got resolved by moving the end-stops over from "polling" to "interrupt" based. Under the "Signal timing diagram" from https://www.antclabs.com/bltouch you can see how short the pulse is (5-10ms).

On the forums there are people claiming that the bl-touch is working like, https://forum.repetier.com/discussion/2477/bltouch-configuration-on-makergear-m2-with-rambo but from his testing section that is clearly one of the clones with a retracted=on and pushed_out=off logic and not the smart sensor with a pulse signal.

Mine is connected to "Z-Min" and I tried V0.92.9 V1.0.2 and V1.0.3dev all with the same simptom.

repetier, 2018-05-31:
Interrupt based pin signals are coming in V2. So for now you need a solution that the timing with pulling is no problem. Firmware tests signal on each step so it depends mainly in the z probe speed. If you use a spindle with 2560 steps per mm it is problem, more on a delta with 80 steps per mm. Also this is already only 12.5ms per step at 1mm/s. At 2mm/s it is already 6.25mm/s. What is your steps per mm and z probe speed. As you see even with coarse resolution it should be no problem to reach the timing.

cor-mt, 2018-05-31:
That's it, working 100% at 20mm/s, will now start to play with the rest of auto calibration, tnx. Maybe a warning should be added for bltouch to the Z-Probing documentation about min speed and making sure eeprom is not overriding it?

That's probably why some people state that bl-touch is working in marlin but not repetier (point). If you want to start save and slow you will never get it working...

gregbiv, 2018-10-07:
Hello everyone! @repetier, I am facing the same issue in my setup. Endstop via bltouch works flawlessly, however, the Z-probing feature doesn't work as expected. 

After successful homing, I do G31 - it says the probe is Low and after deploying the probe by command M340 P0 S700 and I am touching it by hand the G31 - it shows state High.

But one way or another, when I do G30 it ignores bltouch probe and pushing through my print bed and fails when bltouch entering the alarm mode.

I need some advice, what's going on there.

gregbiv, 2018-10-13:
Looks like I've figured out what was the problem. I flushed all settings except what I need for probing feature and found something interesting in logs. It worked on default settings, but when I set offset it went through heating bed again. The reason for it - it couldn't move to a probing point because of offset (the calculated X was negative) but since it got already a signal to move to Z -200, it tried to make a hole in the bed.

Maybe in case of error during the probing, we should halt the printer?