repetier/RepetierFirmware #794: Is there a way to read the fan speed?

luxarts, 2018-06-06:
I'm working on an intelligent system and I need to read the current value of the layer fan speed (pwm value or percentage). Is there a way to get it? Can be implemented a M command to return this value?

Thanks!

luc-github, 2018-06-06:
M408 give the info in JSON format

repetier, 2018-06-06:
Also firmware reports every change with something like
FanSpeed:89
in the output.

luxarts, 2018-06-06:
Thanks. I didn't know the M408 command.

luxarts, 2018-06-06:
I tried the M408 command but the printer throw "Unknown command:M408". Tried using S0 to S5 as parameter but still doesn't work. I'm using the ARM firmware version 1.0.2.

repetier, 2018-06-06:
You need to check the option to add json output, otherwise the command is not included.

luxarts, 2018-06-07:
Thanks, I'll try tomorrow!
Can I ask you another question?
Is there a way to get the current print status (while printing from SD card) using a command?
Btw, I don't know if you remember but I'm adding the support for TFT LCD display for Due. I don't completely understand the syntax you use for style the menus on the LCD 128x64 so I'm hard coding the pages.

repetier, 2018-06-08:
For sd print all you can do is query file position with the proper M code, see repetier.ino for the number, something in the twenties.

Firmware has internally some flags for print state, but if you parse M408 you should get them as well I think.

The menu system is difficult with teh structures. For V2 I'm thinking about also hard coding it as it makes later changes much easier.

closed 2018-06-08 by luxarts