repetier/RepetierFirmware #795: spare i/o's?

emetss, 2018-06-07:
Hi all,
I have an Arduino mega 2560 with ramps 1.4 and a repetier firmware for a 3d printer. I want to add some features to control some additional things (additional fans, and so on). Do I have spare (free) I/Os, which i can use? Also what's the mapping between the mega 2560 and the ramps 1.4 ? I should "define" the pins as their numbers in the arduino but it's not like that. They are defined with different numbers.

Thanks

repetier, 2018-06-08:
RAMPS uses the same arduin pin numbers. In reprap.org wiki you have a pinout whcih pin has whcih number. In general you can use any pin with any function ifthere are no special requirements. For analog inputs you always need a analog pin of course. In config tool just enter the digital pin number you used instead.

emetss, 2018-06-08:
Many thanks for your reply. I'm still confused. I have protos 3d printer which has SD and LCD features used. I don't have the schematics but I have the Firmware. I searched in the internet and found the attached pictures for the schematics. You see as the pictures tell that AIN0 is D54 ?. Also I looked in the Firmware "reptier" for the Protos and under pins.h I found the following:
#define RAMPS_V_1_3
// #define RAMPS_V_1_0

#ifdef RAMPS_V_1_3

#define ORIG_X_STEP_PIN         54
#define ORIG_X_DIR_PIN          55
#define ORIG_X_ENABLE_PIN       38
#define ORIG_X_MIN_PIN          3
#define ORIG_X_MAX_PIN          -1  //2 ----------------------------------------------------------------------------------------------------------

#define ORIG_Y_STEP_PIN         60
#define ORIG_Y_DIR_PIN          61
#define ORIG_Y_ENABLE_PIN       56
#define ORIG_Y_MIN_PIN          14
#define ORIG_Y_MAX_PIN          -1  //15 ---------------------------------------------------------------------------------------------------------

#define ORIG_Z_STEP_PIN         46
#define ORIG_Z_DIR_PIN          48
#define ORIG_Z_ENABLE_PIN       62
#define ORIG_Z_MIN_PIN          18
#define ORIG_Z_MAX_PIN          -1  //19 ---------------------------------------------------------------------------------------------------------

#define ORIG_E0_STEP_PIN         26
#define ORIG_E0_DIR_PIN          28
#define ORIG_E0_ENABLE_PIN       24

#define ORIG_E1_STEP_PIN         36
#define ORIG_E1_DIR_PIN          34
#define ORIG_E1_ENABLE_PIN       30

#define SDPOWER            -1
#define SDSS               53
#define SDCARDDETECT 	    49

#define LED_PIN            13
#define ORIG_FAN_PIN       -1  //8 ---------------------------------------------------------------------------------------------------------------  
#define PS_ON_PIN          12

#define HEATER_0_PIN       10  // e0->t0 --------------------------------------------------------------------------------------------------------- 
#define HEATER_1_PIN       9   // e1->t18   // 8  //hb->t2
#define HEATER_2_PIN       8   // hb->t29   // 9  //e1->t1
#define TEMP_0_PIN         13  // t0
#define TEMP_1_PIN         14  // t1
#define TEMP_2_PIN         15  // t2
#define E0_PINS ORIG_E0_STEP_PIN,ORIG_E0_DIR_PIN,ORIG_E0_ENABLE_PIN,
#define E1_PINS ORIG_E1_STEP_PIN,ORIG_E1_DIR_PIN,ORIG_E1_ENABLE_PIN,

My problem is that I don't know how to read these numbers. Does -1 mean that the pin is not used? How about the other numbers? And Also very important question: How to find and unused pin (digital input in my case) and how to choose it.

Many many thanks in advance

[image]
[image]

repetier, 2018-06-09:
For ramps better use this to see pin numbers:
https://reprap.org/wiki/File:Arduinomega1-4connectors.png

AIN0 is analog pin 0 if used as analog pin and D54 = pin 54 if used as digital pin. That is all.
-1 is not defined as pin number, so it would do nothing.

emetss, 2018-06-13:
Thanks for your kind reply. I checked the schematics you sent to me. There're still some numbers that can't be seen in the schematics. From the definitions below (in the repetier Firmware) I could only find the following pins: D3, D2, D14, D15, D18, D19, D53, D9, D12, A13, A14
Could you tell me where to find D54, D55, D38, D60, D61, D56, D46, D48, D62, D26, D28, D24, D13, D12 in the schematics?

// For RAMPS 1.4
#define X_STEP_PIN         54
#define X_DIR_PIN          55
#define X_ENABLE_PIN       38
#define X_MIN_PIN           3
#define X_MAX_PIN           2

#define Y_STEP_PIN         60
#define Y_DIR_PIN          61
#define Y_ENABLE_PIN       56
#define Y_MIN_PIN          14
#define Y_MAX_PIN          15

#define Z_STEP_PIN         46
#define Z_DIR_PIN          48
#define Z_ENABLE_PIN       62
#define Z_MIN_PIN          18
#define Z_MAX_PIN          19

#define E_STEP_PIN         26
#define E_DIR_PIN          28
#define E_ENABLE_PIN       24

#define SDPOWER            -1
#define SDSS               53
#define LED_PIN            13

#define FAN_PIN            9

#define PS_ON_PIN          12
#define KILL_PIN           -1

#define HEATER_0_PIN       10
#define HEATER_1_PIN       8
#define TEMP_0_PIN          13   // ANALOG NUMBERING
#define TEMP_1_PIN          14   // ANALOG NUMBERING

Many thanks in advance for your kind reply

repetier, 2018-06-13:
Hi,

>:

> Thanks for your kind reply. I checked the schematics you sent to me.
> There're still some numbers that can't be seen in the schematics. From the
> definitions below (in the repetier Firmware) I could [...]
As you see these are hard wired to components like stepper drivers and
therefor do not appear in the external list. These are preset with the
symbolic definition in pins.h as you already found out.

Yours
Roalnd Littwin

> // For RAMPS 1.4
> #define X_STEP_PIN 54
> #define X_DIR_PIN 55
> #define X_ENABLE_PIN 38
> #define X_MIN_PIN 3
> #define X_MAX_PIN 2
>
> #define Y_STEP_PIN 60
> #define Y_DIR_PIN 61
> #define Y_ENABLE_PIN 56 [...]

-- 
Mit freundlichen Grüßen

Roland Littwin

----------------------------------------------
Hot-World GmbH & Co. KG
Knickelsdorf 42
47877 Willich
Deutschland

Tel.: 02154-606518
Fax: 02154-606519

Geschäftsführer: Marcus Littwin
Handelsregister: Amtsgericht Krefeld HRA 5325
Umsatzsteuer-Identifikationsnummer: DE213748632
Persönlich haftender Gesellschafter: Hot-World Medien GmbH mit Sitz in
Willich
Handelsregister: Amtsgericht Krefeld HRB 10800

repetier, 2018-06-13:
Good luck. Do not block interrupts for longer communications and send data only in main thread not from an interrupt. Should work well if done right.

repetier, 2018-06-15:
No that would be enough except M999 already exists to resume from fatal errors.

The better solution would be to do all this with our events. There is a event for M codes and also for init.  Defining pin numbers in pin.h is not really required. You can also define it in your events or directly use your number. See teh sample event systems in the github repository. For the felix printer I use events for all extensions of the printer including M codes.

tecan, 2018-07-24:
https://github.com/tecan/PCF8574-SWire-Arduino.git  adds 8 new i/o ports
https://github.com/tecan/Adafruit_SSD1306-SWire-Arduino

both have the ability to use any of the digital io pins for i2c on arduino mega and due

repetier, 2018-07-25:
But you can not address them like normal io ports, so you have to modify the code changing these pins.