repetier/RepetierFirmware #796: Format of EEPROM settings on SD card

luxarts, 2018-06-15:
Hello!
I'm trying to gete the values of EEPROM from the file 'eeprom.bin' created in the SD card but I can't figure out how the data it's formated. I checked the code but I don't completly understand how the values are saved in the file.
When I open it with a text editor the file has the data splited in 4 hex chars (that means 2 bytes) and with a white space in the middle. EJ: FFFF 0123
How can I "decode" it?

Thanks!

repetier, 2018-06-15:
It is not formatted. It is just a 1:1 copy of what would have been written to the first 4kb in eeprom. So format is exactly the same as it would be in eeprom. So check in eeprom.h where the different parts are stored. At the beginning there is also the eeprom mode and checksum to verify it is a valid copy.

luxarts, 2018-06-15:
I thought that but as an example I set the value for "steps per mm" to 123, save into the eeprom file and opened with a text editor. I searched for "007B" (123 in hex) in the whole file and didn't find anything. I know that the steps per mm value it's a float and it uses more than one byte.

repetier, 2018-06-15:
Bad example. Steps per mm are flats so you need to search for binary representation of 123 and not for integer 123. Best check with a good hex editor that also shows values as float. In eeprom.h you see for these values at which address to search.

luxarts, 2018-06-15:
I made an short javascript program to translate from float to hex and it worked. The problem was that I was reading left to right but the format of float in binary it's right to left.
You learn something new every day.

Thanks for the reply!

closed 2018-06-15 by luxarts