repetier/RepetierFirmware #797: Files on SD card not visible v1.0.2

svs072477, 2018-06-22:
Repetier v 1.0.2
Board RuRamps4D + RepRapDiscount Fullgraphic with SD card reader
The presence of the sd card is determined, but the files are not visible.
On Repetier 1.0.1 its OK

repetier, 2018-06-23:
Where are they invisible? In display or in host software when querying the filenames?

Any error messages in log?

For me it is working, but maybe the 50MHz are too much we use to query sd card. In SDCard.cpp in initsd you see

    if (!fat.begin(SDSS, SD_SCK_MHZ(50))) {

please try a slower frequency there. Maybe I need to reduce this. Please report if it helps and what value works for you.

Nibbels, 2018-06-23:
One question on this: How will SD_SCK_MHZ scale on an Arduino Mega 2560?

I should never reach this speed by the nature of the lower CPU Clock, I guess?

svs072477, 2018-06-23:
SD init fail
SD initialization failed.
Do not reformat the card!
Is the card correctly inserted?
Is chipSelect set to the correct value?
Does another SPI device need to be disabled?
Is there a wiring/soldering problem?
errorCode: 48
I reduce  SD_SCK_MHZ to 20, 10
but it not help
RuRAMPS4D work with Arduino Due

repetier, 2018-06-24:
I think it will scale to highest possible rate. On Mega 2560 this is 8 or 4MHz. Due with 84MHz base frequency can go higher.

Can you test 4 or 2mhz as well. Especially on longer cables you need lower frequency to get it working.

svs072477, 2018-06-24:
With SD_SCK_MHZ(4) it work, thanks

Nibbels, 2018-06-24:
Nice to know!

And now I remember:
I had one card that showed me errorCode 48 and 67 (seemd to be random).
I might have been at the edge of function using this card. And the possible solution to get it working was cleaning the contact plates of the sd-card.

https://www.youtube.com/watch?v=_PKHhuXuy28&feature=youtu.be&t=3m27s
After I wiped of some dirt it worked.  Well this might be just luck and not the solution overall but Error 48 as well as Error 67 might seem to be caused by a communication problem(?)
Errorcode 32 is quite clear. That is SD-Timeout. It appears whenever I put in an empty cardholder for microSDs.

[Ok, if I get this error in the future I will probably switch to a lower frequency at second try.]

Thx @svs072477 for your testing and reporting!!

Greetings

repetier, 2018-06-24:
Ok, will lower default to 4Mhz for next update. Still much higher then baud rate so should not be a problem to fill buffers with that, but work on more devices I guess.

closed 2018-06-24 by repetier

luxarts, 2018-07-03:
I have the same issue. Using 4MHz still doesn't work (errorCode: 32). Using 1.0.1 works fine so isn't a wiring problem.

repetier, 2018-07-03:
Did you reduce both speeds? It gets initialized 2 times, I guess because some cards did not work otherwise. Should look like this:

```    HAL::pingWatchdog();
    HAL::delayMilliseconds(50); // wait for stabilization of contacts, bootup ...
    fat.begin(SDSS, SD_SCK_MHZ(4)); // dummy init of SD_CARD
    HAL::delayMilliseconds(50);       // wait for init end
    HAL::pingWatchdog();
    /*if(dir[0].isOpen())
        dir[0].close();*/
    if (!fat.begin(SDSS, SD_SCK_MHZ(4))) {```

repetier, 2018-07-03:
Also what board are you using? Does it use hardware SPI? I remember someone telling me his board had software SPI for sd card which seems not to work.

luxarts, 2018-07-04:
Yes, I changed both lines (dummy and if statement).
I don't know if I told you that but I'm using software SPI, because I'm using Arduino Due + RAMPS 1.4 modified as the Smart RAMPS and using the Full Graphics display which have the SD pins on the MEGA SPI pins (D52, D50, D51 and D53 for SS).

repetier, 2018-07-06:
What board definition do you use that it defines software SPI for sdfat in old version. Could not find it.

Anyhow seems that software spi is at least possible, see this example:
https://github.com/greiman/SdFat/blob/master/examples/SoftwareSpi/SoftwareSpi.ino

You need to define the sd driver manually and modify a config script to use it just as the example tells you to do. The file is not visible in arduino ide, you need to find it in the file browser. Arduino will only show root folder content. But with that slight modification it should work.

Sanitar777, 2018-08-02:
Hello repetier.
Have DUE + RADDS + RepRapDiscount Smart Controller connected via an adapter, as Original RADDS LCD Display. Had the same error ...

SD init fail
SD initialization failed.
Do not reformat the card!
Is the card correctly inserted?
Is chipSelect set to the correct value?
Does another SPI device need to be disabled?
Is there a wiring/soldering problem?
errorCode: 32

After changing the value SD_SCK_MHZ to 2 everything worked perfectly. 
Thank you. Good job.

luxarts, 2018-08-02:
Hi  @Sanitar777,

I will try with 2MHz in next days. I had to rollback to a previous version.
Thanks for sharing!

Nibbels, 2018-12-31:
Maybe this is a good hint for this old topic here:
Greiman posted some new commits.

1.0.12 Fix iinitialization for older cards
https://github.com/greiman/SdFat/commit/356c5e417c2a06e7853dedef2580857dcd3d2ae0

1.0.8 Fix allocation algorithm
https://github.com/greiman/SdFat/commit/f0c65da00060fcc8c4ac9667d29b2d46557de42e

Greetings

Nibbels, 2019-01-07:
Possible (not that much) work again. @greiman seems to be quite busy these days.
🥇 

1.0.13 Avoid MISO missing pull-up problem
https://github.com/greiman/SdFat/commit/37c1f2606b849e16b9a30d2a0c4b913cefcae287

1.0.14 Improved card re-initialization
https://github.com/greiman/SdFat/commit/60864e38aaa4f319dad5ecc1a912bc97fd12f518

Greetings

repetier, 2019-01-07:
Will be in next update