repetier/RepetierFirmware #800: Wrong filename in SD list

luxarts, 2018-06-27:
Hello! I tried to find a solution to this problem looking at the forums but I didn't find it.
I'm using Repetier 1.0.1 in a ArduinoDue and FullGraphics display.

The problem is that some files are displayed with a strange name.
I copied a file multiple times and changed the name adding 1 character from 10 to 15 to see when this error ocurrs but for my surprise, the 14 characters file (really 14 from name + .gcode =20) is displayed with error but with 15 characters don't.

This is what I get in the display 
[image]

And this is the SD card
[image]

Thanks!

repetier, 2018-06-27:
The strange filename is filename in 8.3 notation. This is one of the reasons 1.0.2 now uses the new sdfat library. That should handle filenames up to 26 chars (ascii only) correctly.

None the less, apart from showing this as 8.3 it should still work.

luxarts, 2018-06-27:
Oh, so I'll update to 1.0.2 then.
Thanks!

closed 2018-06-27 by luxarts