repetier/RepetierFirmware #801: G28 homing command wrong direction

DvDriel, 2018-06-28:
I am trying to properly configure the firmware for a delta printer, but the g28 homing command makes the carriages move in the wrong direction (down instead of up). I tried to find post of similar issues, but could not find anything that worked for me.

Everything seems to work ok, G1 commands work as expected, m119 returns the right states of the endstops.

Changing HOME_DIR doesn't make any difference, both for 1 and -1 the homing goes down. EEPROM_MODE is on 0

#define X_HOME_DIR 1
#define Y_HOME_DIR 1
#define Z_HOME_DIR 1

 What could be the problem?

DvDriel, 2018-06-28:
I got some progression by inverting the X, Y and Z axis. The g28 now homes in the right direction, but ignores the endstops? m119 still gives the expected results.

repetier, 2018-06-28:
Do you use x,y,z max endstops? Deltas need them as max endstop not min endstop.

DvDriel, 2018-06-28:
Okay, I now changed the end stop stops to max ones, but they are still being ignored, it seems like the fw thinks they are inactive or something. Endstops give the right message when I enable their debugging option via the LCD menu. Something I noticed is that the max endstop says: off in the move menu.

#define ENDSTOP_PULLUP_X_MIN false
#define ENDSTOP_PULLUP_Y_MIN false
#define ENDSTOP_PULLUP_Z_MIN false
#define ENDSTOP_PULLUP_X_MAX false
#define ENDSTOP_PULLUP_Y_MAX false
#define ENDSTOP_PULLUP_Z_MAX false

// Set to true to invert the logic of the endstops
#define ENDSTOP_X_MIN_INVERTING false
#define ENDSTOP_Y_MIN_INVERTING false
#define ENDSTOP_Z_MIN_INVERTING false
#define ENDSTOP_X_MAX_INVERTING true
#define ENDSTOP_Y_MAX_INVERTING true
#define ENDSTOP_Z_MAX_INVERTING true

// Set the values true where you have a hardware endstop. The Pin number is taken from pins.h.

#define MIN_HARDWARE_ENDSTOP_X false
#define MIN_HARDWARE_ENDSTOP_Y false
#define MIN_HARDWARE_ENDSTOP_Z false
#define MAX_HARDWARE_ENDSTOP_X true
#define MAX_HARDWARE_ENDSTOP_Y true
#define MAX_HARDWARE_ENDSTOP_Z true

#define INVERT_X_DIR 0
#define INVERT_X2_DIR 1
#define INVERT_Y_DIR 0
#define INVERT_Y2_DIR 1
#define INVERT_Z_DIR 0
#define INVERT_Z2_DIR 1
#define INVERT_Z3_DIR 1
#define INVERT_Z4_DIR 1

//// ENDSTOP SETTINGS:
// Sets direction of endstops when homing; 1=MAX, -1=MIN
#define X_HOME_DIR 1
#define Y_HOME_DIR 1
#define Z_HOME_DIR 1

and also enabled:
#define ALWAYS_CHECK_ENDSTOPS 1

repetier, 2018-06-28:
Important test is always if M119 reports 3 max end stops correctly and if they change if triggered. Also run tests always in a host with open console/log so you see error messages.

DvDriel, 2018-09-12:
I did as you said, but I'm still faced with the same problem. M119 command works as expected, it returns 3 MAX endstops that all report a H status when triggered. When using a G1 command to push the carriages against the endstops the endstops are triggered and the movement stops. using G28 however, the carriages move upwards, but when they hit the endstop the movement continues. What could be wrong?