repetier/RepetierFirmware #806: feature request ( version 2 ) : custom kinematics

ekaggrat, 2018-07-17:
i would be nice if it is as easy as smoothie to put a custom kinematics support in version 2. I am playing around with a gus and it would be nice to be able to change the kinematics as easily it can be done in smoothie.. Right now i am using marlin with modified delta equations but it is a PITA to figure out all the linked functions .

repetier, 2018-07-17:
That is how we do it by design. Each kinematics has its own class and that should be all you need to modify. Currently I‘m working on the delta kinematics there. I mean the whole version becomes a modular set of exchangeable functions making it as easy as possible to replace components. I think this is the future of a flexible firmware concept.

ekaggrat, 2018-07-17:
thats cool .. any date when the beta testing will be available?

repetier, 2018-07-17:
It is in dev2 branch already but would not consider it beta.  Functions might still change if i see a need. Hope to get a good set of functions by end of august, but that is no promise. Other problems may easily change dates here.

whirlybomber, 2019-02-11:
I've just got a DoBot Magician with the integrated linear rail. The DoBot is using Repetier firmware to 3D print when it is running by itself, so I came here to look at the source files to see how difficult it would be to write a new HAL/interpreter so that the arm would do the X and Z movement, while the Y direction would now be handled by the L direction of the linear rail (so it would be a cylindrical coordinate system).
I want to get something moving in the next ~3 weeks, so I'm not sure that I should be looking at v2 firmware, as most posts suggest that even though it is simpler to make these sorts of updates to, it is not compiling and running yet.
So I looked in the master branch, and the files that look like I should be working on are motion.h and hal.h, with the hal.h file looking like the most likely one to need modifying.
I've looked at the wiki as well, and that seems to be fairly out of date.
So am I looking in the right place (files) to make changes in, or have I missed the plot entirely?

repetier, 2019-02-11:
V2 is working good if you have the matching hardware. It compiles without problems if you follow the docs. I think it is much easier to integrate in V2 then in V1 if you have a Arduino Due based board which is the only supported one for now. If you need to control the original board it might get complicated as I assume it is no Arduino board? HAL for V1 and V2 is currently not much different so nearly same work. V2 has hardware pwm support.

V1 is development branch where also updates take care. Master is only updated when I think a new dev version is ripe to get published. Will do this quite soon.

In V2 you just add a new printer type (cpp/h file) and modify the include rule and you are more or less  done. All printer type dependent code is there. In V1 the code is quite distributed over multiple files. It i snot only the mapping to motor positions that needs changes, also homing and autoleveling might need special handling.

whirlybomber, 2019-02-12:
Thanks for the quick reply.
I've been looking more into the software and am somewhat confused by the whole situation with the DoBot.
As far as I can tell, the Repetier firmware works by outputting pulse / direction to GPIO pins on whatever microcontroller it is running on. Whereas as best I can tell, the DoBot drivers for joint 1, 2, 3 and 4 are driven from a FPGA that connects to the microcontroller in the bottom of the DoBot - which should not work as far as I understand it. The best description I could find is [here](https://github.com/maxosprojects/open-dobot/wiki/3.-FPGA-version-system-diagram).
I downloaded the DoBot software and manual from the [DoBot website](https://www.dobot.cc/downloadcenter.html).
I looked through the manual and confirmed that all the versions of the software I was using were the same as listed in the manual. One strange thing I did notice was the revision number listed against the RepetierHost when the DoBot Studio applies the 3D printing firmware. The dialogue box says that it is installing RepetierHost v4.6.0 as the firmware. Which we know is incorrect as the firmware should be a file RepetierFirmware v.blah
Also, the version of RepetierHost it actually loads is the RepetierHost v1.0.6.
So I assume that the RepetierFirmware that is being loaded is v4.6.0 but I cannot find that in the DoBot folder from the downloaded software. The only files I could find that might be the repetier firmware are not supported for download so I can't link them but are called RepetierHost.hex and RepetierHostExtender.dll
So do you know what firmware is being loaded onto the DoBot when it says it is uploading RepetierHost v4.6.0? Did DoBot fork the Repetier project and make their own version?
Thanks for your help.

repetier, 2019-02-12:
Looks like they use RAMPS board so the 8 bit version is used here. I see no references to repetier. What I understand is that here the FPGA controls the arm and the arduino just sends commands to the FPGA. That means V2 firmware will not run and V1 will also be of no benefit as we control step/direction of servos directly to sync everything. Syncing moves with an external controller is a completely different approach and requires a different solution and very good knowledge of the fpga working. I mean it is no problem to send target positions to an arm. But if you do 3d printing timing is very important to be at every ms where you think the arm is. This might be possible but the logic needs big adaptions in that case. How difficult it is depends on how the control works. e.g. can you queue moves so next part is started in time, so you only have to fill the queue? That would be easy and you just need to send the moves and wait for a free slot. Maybe do some speed limitations so the arm does be able to move.

whirlybomber, 2019-02-12:
Hello again repetier. Yeah, I don't know exactly what firmware they load to replace the Dobot firmware before you run 3D printing on the Magician, but the manual (and the firmware loader) both say they are loading repetierhost software to the controller. It even has a chapter heading in the manual that says "5.9.2 Operating Repetier Host. Repeiter Host has been built into DobotStudio. After burning 3D printing firmware, the Repetier Host page will be displayed automatically."
So maybe they aren't actually using any repetier firmware in the machine. I dunno.
But you are correct in that there is a FPGA in the system board and that it does control the arm directly. It also has a series of encoders on each of the joints as well, so that the arm at least is a closed loop control system. Having looked closer it seems that the linear rail doesn't have an encoder, just and end stop, and it is plugged into the microcontroller (arduino in the picture I linked) so it would be even harder to coordinate driving it accurately in association with the FPGA GPIO's.
Unfortunately since DoBot don't tell us a whole lot about the settings of the system, I doubt it will be possible for me to reverse engineer enough of they system to load anything on it.
Of course that doesn't stop us implementing a 3D printer by using software to turn G-code into instructions in the API to make it run. Just hoping that it would be accurate enough to be useful.
eh.
maybe I'll just suggest getting a beaglebone / machinekit board and using that.......
Thanks for your help.