repetier/RepetierFirmware #808: M204

MrJabu, 2018-07-25:
hello,

currently running v1.0.2.

the issue i am having is that i can not get M204 to set the PID for the hotend.

i am using code M204 X24.51 Y3.54 Z63.71 and entering it within Repetier-Host but i get no response and my PID's are the still the old values.

MrJabu, 2018-07-29:
is there anyway to do this in GCode? i wanted to add different settings to the startup gcode depending if i print ABS or PLA.

repetier, 2018-07-29:
- M204 - Set PID parameter X => Kp Y => Ki Z => Kd S<extruder> Default is current extruder. NUM_EXTRUDER=Heated bed

This will not change the values shown in eeprom, only internal values and every command changing eeprom will reset them, but these are not in a print so using this in gcode should work.

MrJabu, 2018-07-29:
I tried using that command but didn't see any change in eeprom. Makes sense now. So it actually was changing the parameters?