repetier/RepetierFirmware #811: "Heater decoupled"

leoheck, 2018-08-06:
I am having this "Heater decoupled" very often.
Using Repetier-Firmware v0.92.9

Tried to print 3 times right now. The temperatures graphs are just fine. And suddenly I have this _Heater decoupled_ error which messes the printer job. 

Maybe it is a problem with the thermistor/cables but they look good. I just measured the thersmistor which was showing a steady ~102K today.

[image]

I found this message, which may be related.

[image]

Any idea on what I can try to fix this thing?

leoheck, 2018-08-06:
Maybe it is related with #618

leoheck, 2018-08-06:
Trying it again after upgrade the Repetier-Server from v0.90.1 to v0.90.2
All seems to be good but it suddenly stops with the "Heater decoupled" issue.

**Any ideas on why this happens?**
How can I fix it?

[image]

[image]

leoheck, 2018-08-06:
Printing log from the server.
Is it possible to be something related with the model/gcode?

```
15:59:41.868: FIRMWARE_NAME:Repetier_0.92.9 FIRMWARE_URL:https://github.com/repetier/Repetier-Firmware/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mendel EXTRUDER_COUNT:1 REPETIER_PROTOCOL:3
15:59:41.873: Printed filament:708.66m Printing time:16 days 14 hours 50 min
15:59:41.876: PrinterMode:FFF
15:59:41.881: Info:No case lights
15:59:41.884: X:0.00 Y:0.00 Z:0.000 E:0.0000
15:59:41.887: DebugLevel:6
15:59:41.890: Config:Baudrate:115200
15:59:41.893: Config:InputBuffer:127
15:59:41.895: Config:NumExtruder:1
15:59:41.896: Config:MixingExtruder:0
15:59:41.898: Config:HeatedBed:1
15:59:41.900: Config:SDCard:1
15:59:41.901: Config:Fan:1
15:59:41.901: Config:Fan2:0
15:59:41.903: Config:LCD:1
15:59:41.906: Config:SoftwarePowerSwitch:1
15:59:41.906: Config:XHomeDir:-1
15:59:41.908: Config:YHomeDir:-1
15:59:41.911: Config:ZHomeDir:-1
15:59:41.911: Config:SupportG10G11:1
15:59:41.918: Config:SupportLocalFilamentchange:1
15:59:41.918: Config:CaseLights:0
15:59:41.925: Config:ZProbe:0
15:59:41.925: Config:Autolevel:1
15:59:41.926: Config:EEPROM:1
15:59:41.926: Config:PrintlineCache:16
15:59:41.926: Config:JerkXY:20.00
15:59:41.928: Config:JerkZ:0.30
15:59:41.931: Config:RetractionLength:3.00
15:59:41.933: Config:RetractionLongLength:13.00
15:59:41.936: Config:RetractionSpeed:40.00
15:59:41.939: Config:RetractionZLift:0.00
15:59:41.945: Config:RetractionUndoExtraLength:0.00
15:59:41.945: Config:RetractionUndoExtraLongLength:0.00
15:59:41.950: Config:RetractionUndoSpeed:20.00
15:59:41.951: Config:XMin:0.00
15:59:41.953: Config:YMin:0.00
15:59:41.953: Config:ZMin:0.00
15:59:41.956: Config:XMax:150.00
15:59:41.956: Config:YMax:185.00
15:59:41.958: Config:ZMax:160.00
15:59:41.961: Config:XSize:150.00
15:59:41.961: Config:YSize:185.00
15:59:41.964: Config:ZSize:160.00
15:59:41.966: Config:XPrintAccel:500.00
15:59:41.969: Config:YPrintAccel:1000.00
15:59:41.971: Config:ZPrintAccel:1.00
15:59:41.972: Config:XTravelAccel:1000.00
15:59:41.975: Config:YTravelAccel:1000.00
15:59:41.978: Config:ZTravelAccel:15.00
15:59:41.980: Config:PrinterType:Cartesian
15:59:41.983: Config:MaxBedTemp:120
15:59:41.986: Config:Extr.1:Jerk:20.00
15:59:41.988: Config:Extr.1:MaxSpeed:50.00
15:59:41.991: Config:Extr.1:Acceleration:5000.00
15:59:41.994: Config:Extr.1:Diameter:0.00
15:59:41.996: Config:Extr.1:MaxTemp:275
15:59:41.999: Unknown command:N11 M539 S1```

repetier, 2018-08-07:
Ok, I see where your decoupling comes from I guess. Your temperature graph is very bad. I see temperatures swing at 10°C within seconds. This is not correct and a good curve is much smoother normally. It swings within 0.5°C regarding measurement errors and not that much. So find out what is causing the noise. Firmware already smoothes inputs but it seems not enough.

At first check if the cable for temperature is twisted and has no big contact to high power cables like heater or motors. Induction can cause lot of noise here. If they have contact twist these cables as well. That equals the inductive effect to a great amount (which is why network cables are also twisted). 

Also as a test just check temperature when motors/heaters are off. Then it must be a very flat line with minimal swings. If you have already big variation when everything is off the error is something else. Then try a 100K resistor instead of thermistor or test a different thermistor to rule out a defect there.

leoheck, 2018-08-07:
Good thanks for the explanation. So I have some issues here. My thermistor cable is not twisted and it touches the hotend wires all the way long. I will try to make the tests you said and improve my setup.

Also, I am planning to upgrade the firmware to the latest version. Do you know if it can improve something considering issues related to the thermistors? Is it good to apply the latest version?

repetier, 2018-08-08:
Updating will not change anything with the temperature reading. It is pure hardware related that the ADC delivers varying values. Of course 1.0 has many new things and even fixes for other problems so upgrading is always a good thing.

leoheck, 2018-08-20:
Do you have an example of temperature graphs that are actually good? I think i have some problems. 
I set 60oC to the extruder. This is the variation I have. It does not look like you said.

[image]

repetier, 2018-08-21:
I have no available at the moment, but a normal curve has no spiked about 2°C within a few seconds. In fact there even is a filter to remove single spikes, but you seem to have longer periods of changed voltage. Apart from crosstalk from heaters it also can be a defect thermocouple. When they expand during heating they might get tiny breaks due to stress from holding it inside. That also can lead to unstable values. If you have a resistor you can replace thermistor with it. Then it should be a quite stable line as resistance does not change. Also disabling heaters and motors should remove influence like crosstalk and show a stable line if it is from crosstalk.

leoheck, 2018-08-21:
With a short (15 cm) of twisted wire and a 100k resistor, the line is flat.
It was me messing with wires before `57:45`. So my problem my be the thermistor itself since in the last graph 3d printer was it working with motors off and the thermistor wire was far from the other wires.

[image]

repetier, 2018-08-21:
Yes, I think you can now say it comes from thermistor. That flat line is how it looks like. With a thermistor it varies a bit due to temperature changes but it is similar.

leoheck, 2018-08-23:
I am buying a new thermistor, but I still feel that the problem isn't the thermistor. There should be something else. Analysing some prints I took from the temperature curves when the problem occurs, I saw that it always happens 7 minutes after heating the extruder, even changing the 3D model that is being printed.

This happened yesterday while I was testing. I sent the print through the server. 

When I start the printing from the host it works. I am going to test it one more time to garantee that this statement.

[image]

repetier, 2018-08-23:
You can disable decouple testing by setting the timeout to 0. The problem is that the errors you measure are random and mostly do not trigger but under bad conditions it might trigger. Temperatures are controlled by firmware so host or server should not really make a difference if you do exactly the same apart from randomness of temperatures.

Hope it gets better with new thermistor.

repetier, 2018-08-24:
In your case it could also be the variance that triggers. Period is a factor when first heating occurs. E.g. if start was on peak high and control on a peak low the decouple period must be long enough that it is still a temperature rise.

leoheck, 2018-08-24:
Changed the thermistor, finally. Heated bed, extruder and motors, and fan are off. It is currently glued with tape on top of the heated bed.

It is an NTC 3950. Is it good enough?

[image]

leoheck, 2018-08-24:
Just for the sake of comparison, this is my current themperature sensor (I don't know the model, it should be the same as the one I bought). All is off but it still assembled in the extruder.

They look the same! 

[image]

leoheck, 2018-08-24:
Is it possible that I am using a wrong configuration in the firmware? I am sharing it here just in case.
[Configuration.zip](https://github.com/repetier/Repetier-Firmware/files/2320013/Configuration.zip)

Nibbels, 2018-08-26:
Hello leoheck,

I dont remember what heat_manager 3 was. In case you are using PID-Controll, the extruder output (PWM) looks a bit shaky for me. It is 0-1-0-1-0-1-0 instead of a soft line.

If you do an autotune, I would use Classic-PID or pessen integral rule for a hotend.

If you have hardware problems with your electronics sometimes a tiny ceramic capacitor helps alot. I mean some pF or nF to damp away the high frequencys or radio input.

leoheck, 2018-08-27:
Hi @Nibbels, the EXT0_HEAT_MANAGER 3 is the "Dead Time Control"

This is the config that came originally with my 3D printer. So, what is the best out-of-the-box cofiguration here? What is the value for Classical-PID, 0 or 1?

The Repetier-Firmware configuration tool does not have/show me a lot of options, see:
[image]

repetier, 2018-08-27:
He talks about PID calibration using M303

- M303 P<extruder/bed> S<printTemerature> X0 R<Repetitions> C<method>- Auto detect pid values. Use P<NUM_EXTRUDER> for heated bed. X0 saves result in EEPROM. R is number of cycles.
				method 0 = classic, 1 = some overshoot, 2 = no overshoot, 3 = pessen, 4 = Tyreus-Lyben

You need PID Controller for this to make sense. Might work better with these fluctuations you have, also D/P will get wrong during spikes you have the I part that takes longer to follow.

leoheck, 2018-08-27:
@Nibbels I tested some caps in the termistor therminals, near the board. I tested some Values from `22pF` to `0.4uF` but the curves have the same spikes. Nothing has changed at least considering the room temperature, maybe the thermistor works better when it is hotter. I am measuring 15.5℃ (+2℃/-2℃)

[image]

@repetier, you said:
> You need PID Controller for this to make sense

Is it a configuration or a hardware? Can you explain better?

@Nibbels, 
> I dont remember what heat_manager 3 was. In case you are using PID-Controll, the extruder output (PWM) looks a bit shaky for me. It is 0-1-0-1-0-1-0 instead of a soft line.

Sure, tottaly agree. I want to do the autotune to see if it helps. Can you explain this better or reference some tutorial?

leoheck, 2018-08-27:
Oh, I just need to run the command, got it.

@repetier, 
Where should I put the method?
And how much repetitions should I do?

Methods:
- 0 = classic
- 1 = some overshoot
- 2 = no overshoot
- 3 = pessen
- 4 = Tyreus-Lyben

So, right now I am doing this:
```
# M303 P<extruder/bed> S<printTemerature> X0 R<Repetitions>
M303 P0 S200 X0 R8 C
```

leoheck, 2018-08-27:
I got, this. Is it right or good? How do I know?

[image]

repetier, 2018-08-27:
Yes, that is how autocalibration looks like. Copy values to eeprom or if you used X0 it should already be in eeprom.

I remember your experiment with a fixed resistor getting a flat line. Have you tested the line when you have the new thermistor instead connected? I do not mean in extruder just short line. Often it is a good idea to start from a working position adding tiny steps until the error occurs. If it crosses for example later a 220V 50Hz cable that could also cause noise on it. Of course you can also simply unplug the main power. Most boards will still run and report tempertutures.

leoheck, 2018-08-27:
> Have you tested the line when you have the new thermistor instead connected?

Yes, I added it here some comments before, look https://github.com/repetier/Repetier-Firmware/issues/811#issuecomment-415869341

> I do not mean in extruder just short line

I didn't get that, can you explain better? The new temperature sensor was connected in the RAMPs board but it was out of the extruder. Motors were off (Axis/Extruder/Fan).

> If it crosses for example later a 220V 50Hz cable that could also cause noise on it

The thermistor is far from the AC cable. Now it is also far awy from other wires. But I still have the same fluctuation. 

> Of course you can also simply unplug the main power. Most boards will still run and report tempertutures.

Sure, my power supply was off.

There should be something else that is makeing my printer abort the job with the following error. 
[image]

I don't believe that it is due to the thermistor but I am running out of the ideas to test it.

Nibbels, 2018-08-27:
Ok, nice!

Your Ki, Kp and Kd constants look plausible.

Away from that, if you cannot get rid of disortions: At least in germany we have 220V lines with Plus, Ground and Safety Line. In case your Ground is somehow bad, there is some rumor that you should just plug your cable inside the wall turned around. That means you switched plus and ground.
That might, in some specific hardware configurations better your situation. I heard that would make sense at most, when as example your 12V Ground is connected to the green-yellow Safety Line.

LG

leoheck, 2018-08-27:
Cool, autotune the PID was good. It is printing more than 7 minutes :) 

I am using the original thermistor. 

Everything looks good so far
[image]

leoheck, 2018-08-27:
Actually, I am not sure if this is good. The swing seems to be increasing.

[image]

leoheck, 2018-08-27:
Cool, it is getting stable again.

[image]

leoheck, 2018-08-27:
@Nibbels @repetier my test part just finished fine! So, it is working again. I still have to print something that was giving me issues to see if it is ok. Guys, thank you for all the help!!

leoheck, 2018-08-27:
Does this autotune process works with the heated bed?

repetier, 2018-08-28:
Heated bed sometimes works sometimes not, depends on how fast it reacts. Just test it.

Swings in last image are good. Micro swings are look liek only last bit is unstable and that is to expect. Overall swing < 1°C is also very good. With the zoomes temperature axis they look big but looking at the value syou see they are small.

Nibbels, 2018-08-28:
@leoheck in general the different method rules are there to search different "working" sets of Kp Ki and Kd.
The best strategy is that which finds a non-swing but least stable set of constants (Kp Ki and Kd). That is supposed to be the "most reactive one which is stable in all situations".

-> When the temperature rises the hotends/heatbeds temperature line should not overshoot much and go straight into a constant line of temperature.
-> And it should not swing afterwards.
-> And it should not start to swing when you turn on your cooler fan or such. Then your stability is a bit to low.

The following list is my personal order of the methods according my feel of the heaters reactiveness/speed:

reactive hotend ....
3 = pessen integral rule
0 = classic pid (ziegler nichols)
1 = some overshoot
2 = no overshoot
4 = Tyreus-Lyben
... 220v overpowered heatbed

So **I would** try myself from top to bottom for a hotend and from bottom to top for a heated bed and then make pictures like you did to compare the line of rising temperature (step response/Sprungantwort). 
Afterwards use the most pretty one.

_Additional not really needed bonus information: You can decrease an initial overswing by setting #define xxxxxx_PID_INTEGRAL_DRIVE_MAX 230 to a bit lower number on that specific controller. But if you get this number like 5 too low, you will totally waste your controll 🗡 . So be aware that this method of overtune a near perfect set of constants is a bit work.
As example: My hotend had a good response but overswing at the first temperature rise of about 10°K. When I set my xxxxxx_PID_INTEGRAL_DRIVE_MAX 230 to 150 it was only 4°K, But When I set it to xxxxxx_PID_INTEGRAL_DRIVE_MAX 145 the controller did not reach the temperature anymore. So I set it to a number like 150..153_

Greetings

leoheck, 2018-08-29:
Now I understood the `M303` command since the comment about its syntax was a little bit confusing. It will be good to have the line 144 upgraded [here](https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoAVR/Repetier/Repetier.ino).

Improving my extruder a little bit. I will keep my tests here.
```
M303 P0 S200 R12 C3
M500
```

[image]

Before I got this:
```
Classic PID
Kp: 25.98
Ki: 3.19
Kd: 52.83

```
Now it is:
```
Pessen Integral Rule PID
Kp: 16.16
Ki: 1.70
Kd: 57.73
```
**Is it normal to have these differences when running the autotune with different methods?**

leoheck, 2018-08-29:
I coudnt autotune the heatedbed. Should I enable something on firmware to make it work??

repetier, 2018-08-29:
Heated bed is same as extruders. Just set P<numberofextruder> so P1 if you only have one extruder.

Yes, the differences in parameter are normal. It is a trade off between faster converging, overshoot etc.

leoheck, 2018-08-29:
I sent M303 with P1, but it is heating the bed and the extruder at the same time. Is it right?

[image]

repetier, 2018-08-30:
No, it should only heat the bed. Looking at the temperature graph I also see only bed temperature rising, the red extruder line stays at same temperature. Just make sure all heaters are off when you start autotune.

leoheck, 2018-08-30:
Sure, on the 1rst box, only bed temperature was rising. But why the "Output extruder" has increased too?

repetier, 2018-08-30:
I need to see the temperature report in log to say that. Enable ack in log and in printer settings disable removing M105 responses. Then I can say if firmware reports wrong or host interprets wrong.