repetier/RepetierFirmware #816: BLTouch going into alarm mode because printer is measuring z-min

2987ms, 2018-08-26:
Here's a video of my issue: https://youtu.be/9nCiKDn_jS4

My 3D Touch (BLTouch clone) is going into alarm mode because the printer doesn't raise up after measuring z-min. Then it proceeds to try to auto-level, but it fails because the "endstop" is already triggered (in alarm mode). Looking at [Pigeon's tutorial](https://youtu.be/U2zxzn0K_5U?t=514), his printer doesn't even measure z-min.

I have a Monoprice Maker Select v2.1 running Repeteir. [Here is my Configuration.h.](https://pastebin.com/GpSGGBcL)

Either I need the printer to raise up, or even better I need to disable the z-min measurement (since it already has auto-bed leveling enabled, right?).

repetier, 2018-08-27:
You could try setting
#define Z_UP_AFTER_HOME 10
so z goes up after z homing. Hope this is in time.

2987ms, 2018-08-27:
I added that line into my code:
https://pastebin.com/AV5qxApp
However the result was exactly the same. It didn't raise up. Maybe it's not really checking for z-min? I'm not sure, I just assumed it was because it was going to 100,100

2987ms, 2018-09-04:
Bump - Where should I be placing the line `#define Z_UP_AFTER_HOME`? I placed it underneath `#define ENDSTOP_Z_BACK_ON_HOME 0`.

MatthewSerinity, 2018-09-04:
Alright, I've got some more videos to demonstrate the problem better. I switched to v1.01DEV to see if it would make a difference, which it didn't.

This is what it does the majority of the time:
https://youtu.be/Bm0ZEfYB-20

This is what happens if I push down on the buildplate. Notice that it seems to want 3 probes before continuing, but only raises up for two. I changed my `Z_PROBE_REPETITIONS` to 2 to try to combat this, and it didn't affect it:
https://www.youtube.com/watch?v=30zLrzBXHuI

However very rarely, it will only want 2 probes. This has happened 3 times now, once with `Z_PROBE_REPETITIONS` set to 2 and twice with it set to 3:
https://www.youtube.com/watch?v=b5nB6P8dv2o

When it goes through the probing, you can see that it seems to be probing in a diagonal pattern, eventually making it's way a cm or two off of the buildplate (skip to the end):
https://www.youtube.com/watch?v=b5nB6P8dv2o

I would think this means my configuration is wrong, however it goes to 100,100 perfectly fine as specified.

Sending a home-z command makes it probe in the same place (100,100), mess up the same way, and then it goes back to the bottom left corner (rather than continuing with bed leveling).

Here is my Configuration.h file:
https://pastebin.com/NAG7zuLm

MatthewSerinity, 2018-09-04:
Update: I don't know what I did, but I fixed the diagonal-autolevel issue. I just completed my first autolevel, although I had to "cheat" by pushing the bed down on initial z-min. Here is my current configuration.h:

https://pastebin.com/SJf66xRp

MatthewSerinity, 2018-09-05:
It seems that it ends up being triggered too late during z-min probing, does this affect distortion correction? I also found a bug with distortion correction, even when it's enabled via the arduino IDE it's still reported as disabled. Sending ``M323 S1 P1`` fixes that.

I believe the z-min is completely messing this up, and I can't seem to stop it from executing it.

repetier, 2018-09-05:
Distortion correction is only enabled after running G33, before it is normally disabled.

Homing Z doe sonly one probe to find the reference height. If that is wrong distortion correction will still do it's thing but on a wrong level.

Looking the homing video I see this:
1st probe works as it directly goes upwards
second probe does not do this as this is normally then end position. It first runs the finish probe command returning to original offset. During that procedure you get the z triggered and seems to be in an error state. Apart from this it homes ok.

Now when you start probing with bltouch still in error state it reports z triggers and reports the error I saw at the end.

Is that so far correct?

The first question is why does the bl touch reengage the probe and does not keep it up? I see you have
#define Z_PROBE_START_SCRIPT "M340 P0 S700"
#define Z_PROBE_FINISHED_SCRIPT "M340 P0 S1500"

and Z_PROBE_FINISHED_SCRIPT gets run directly as first part. See here code for last move

        PrintLine::moveRelativeDistanceInSteps(0, 0, axisStepsPerMM[Z_AXIS] * 2 * ENDSTOP_Z_BACK_MOVE * Z_HOME_DIR, 0, homingFeedrate[Z_AXIS] / ENDSTOP_Z_RETEST_REDUCTION_FACTOR, true, true);
#if Z_HOME_DIR < 0 && Z_PROBE_PIN == Z_MIN_PIN && FEATURE_Z_PROBE
        Printer::finishProbing();
#endif

as far as I see it this should get called. Have you tested if this works. E.g. probing a single point where you are just for fun using G30 P3 should activate and deactivate it correctly.

The diagonal problem is just having the 3 points not set correctly. First point is the edge and P2/P3 are direction from that point that span a parallelogram. But you seem have to figured this out.

MatthewSerinity, 2018-09-05:
Running G30 P3 results in the following error:
`Changing monitoring state from "Error: Activating z-probe would lead to forbidden xy position: -17.000000, 46.00" to "Offline (Error: Activating z-probe would lead to forbidden xy position: -17.000000, 46.00)"
Connection closed, closing down monitor`
And the printer raises Z by 10mm or so.

However if I "trick" the printer as shown in this video, it probes perfectly fine for the distortion correction:
https://www.youtube.com/watch?v=HakIDpGhaRk
Also note you can see the hot-end push down on the corner of the heatbed.

It doesn't quite home ok, because it takes awhile for it to trigger z-min, leaving z-min measured incorrectly (lower than it should be). When it's in an error state it reports z triggered, yes.

repetier, 2018-09-06:
You need to start G30 where you can activate z probe or you get that error.

>Also note you can see the hot-end push down on the corner of the heatbed.
Sorry, can not really see this. But probe must of course trigger first otherwise the probe is too high. Maybe lowering it 1mm would help here. This might also depend on bed rotation but normally can be covered by enuogh difference between nozzle hit and zprobe trigger.

What I see is that probe is fine as long as it goes up directly after measuring which homing sequence does not do. But it should execute M340 P0 S1500 which is supposed to keep the probe up and would prevent the error. My question  is why does it not work. Or does it work but is too slow so that error state comes first?

repetier, 2018-09-06:
I have an idea that might solve the problem. In finish probe script go to relative mode, go up 1-2mm THEN disable probe, go back down 1mm and go back in absolute mode. That would be like G33 then and should work. You never had a problem when directly going up.

MatthewSerinity, 2018-09-06:
Forgive my ignorance but what would the code be for that?

repetier, 2018-09-07:
"G91\nG1 Z2\nM340 P0 S1500\nG1 Z-2\G90"
would be the sequence. Maybe a G4 S1 after M340 if disabling is too slow.

MatthewSerinity, 2018-09-07:
@repetier The printer didn't seem to like that code at all:
https://www.youtube.com/watch?v=4wkmvntE3QU

MatthewSerinity, 2018-09-09:
I purchased another BLTouch (this time an authentic one) and it resolved my issue.

closed 2018-09-09 by MatthewSerinity

jtkDvlp, 2020-06-15:
Got the same issue and solved with following "Probe start script". But dont know why its working, it seems the script is execute once at the very beginning of G32 no more time oO

`G91\nG1 Z10\nG90\nM400\nM340 P0 S700`