repetier/RepetierFirmware #819: Add another Probe

skwhite16, 2018-08-29:
I want to add functionality for a second probe. Could run it from the Y min pins. It doesn't need to work with the bed-leveling features, I just need it to probe and return a value. Right now I am using G30 to probe in Z and ideally I could just use G30 to run both probes simultaneously and return both values. But it would also work if I could add another Gcode command (ex G31). How would I go about adding this functionality?

luxarts, 2018-09-11:
Why you need a second probe?

skwhite16, 2018-09-11:
I am not building a traditional 3D printer. I am building a machine to extrude a strip of functional material (electrically conductive PP) onto a injection molded part. My original printer consisted of 2 axes: an X axis that carries the part and a Z axis that moves the extruder up and down to follow the contours of the part as it prints. The parts have enough variation that I need to take measurements and make a slight adjustment to the G-code each for each print, which is what I use the probe for. Now I am trying to parallelize the process so that I can print on two parts at a time. So I still have a single horizontal X axis that is now carrying 2 parts. I have 2 independent vertical axes carrying the extruder (Z and Y) so I can print on both parts simultaneously. The Z and Y have to be able to move independently, due to the variation in the parts. The variation in the horizontal direction is negligible, but the variation in the vertical direction is enough that the Z and Y movements cannot be the same. The reason I want two probes is so that I can take measurements of both parts. So I have been using the Z probe to probe the Z axis, and now I want a Y probe so that I can probe the Y axis. There doesn't seem to be a simple way of doing this with repetier firmware, so I may switch to marlin or reprap firmware.

luxarts, 2018-09-11:
I think it will be the same with Marlin or RepRap. If you just need to return value of the sensor (I mean, 0 or 1) go to `Commands.cpp` file and add the command to the `void Commands::processGCode(GCode *com)` function. Add the code for reading and output there, which basically it's a pin read and output through serial port or display. But if you want to return the value in milimeters I would hard code the distance and do the math as the G30 does.