repetier/RepetierFirmware #820: Layers shifted

leoheck, 2018-08-29:
I was printing more than 20 hours in a row, and I got this. 

Many layers were shifted. 
The parts were firmly fixed on the heated-bed glass.

Do you have any clue about why this happens, and how to fix it?

The previous job was 3 pieces of the same part that is in the center of this image.
It was the 1st time I tried to print so during so many hours in a row and so many parts together.

[image]

Nibbels, 2018-08-29:
I dont think this is a firmware issue.

You normally get this effect, when your acceleration or jerk is too high.
Or possibly when you set your acceleration === 0
Or if you didnt use ramp acceleration
Or if your steppers are to weak. -> Amps to low, Stepper to small,
Or if your rails do block/rub somehow.
Or if some of your cables are broken just a bit.

leoheck, 2018-08-29:
Yes sure. Do you know where I can ask this?

closed 2018-08-29 by leoheck

repetier, 2018-08-30:
Also make sure 
#define ALWAYS_CHECK_ENDSTOPS 0

If you have endstop check enabled a shift can happen if a endstop triggers or gets triggered by crosstalk. This normally shifts away from endstop. But if you have not 2 endstops in x direction it is more likely a physical thing like @Nibbels said. Additional reasons are jerk too high, acceleration too high, travel speed too high. The faster your motor runs the less torque it has.