repetier/RepetierFirmware #822: Cant compile firmware

SplitX1, 2018-09-05:
Arduino: 1.8.6 (Windows 8.1), Board: "Arduino Due (Native USB Port)"

In file included from sketch\Repetier.h:572:0,

                 from sketch\ui.cpp:20:

uiconfig.h:231:31: error: pasting "/* PINK.1, 88, D_RS*/" and "_PORT" does not give a valid preprocessing token

 #define UI_DISPLAY_RS_PIN  8  // PINK.1, 88, D_RS

                               ^

sketch\HAL.h:170:49: note: in definition of macro '_WRITE'

 #define  _WRITE(port, v)   do { if (v) {DIO ##  port ## _PORT -> PIO_SODR = DIO ## port ## _PIN; } else {DIO ##  port ## _PORT->PIO_CODR = DIO ## port ## _PIN; }; } while (0)

                                                 ^

sketch\ui.cpp:743:5: note: in expansion of macro 'WRITE'

     WRITE(UI_DISPLAY_RS_PIN, rs);

     ^

sketch\ui.cpp:743:11: note: in expansion of macro 'UI_DISPLAY_RS_PIN'

     WRITE(UI_DISPLAY_RS_PIN, rs);

           ^

uiconfig.h:231:31: error: pasting "/* PINK.1, 88, D_RS*/" and "_PIN" does not give a valid preprocessing token

 #define UI_DISPLAY_RS_PIN  8  // PINK.1, 88, D_RS

                               ^

sketch\HAL.h:170:84: note: in definition of macro '_WRITE'

 #define  _WRITE(port, v)   do { if (v) {DIO ##  port ## _PORT -> PIO_SODR = DIO ## port ## _PIN; } else {DIO ##  port ## _PORT->PIO_CODR = DIO ## port ## _PIN; }; } while (0)

                                                                                    ^

sketch\ui.cpp:743:5: note: in expansion of macro 'WRITE'

     WRITE(UI_DISPLAY_RS_PIN, rs);

     ^

sketch\ui.cpp:743:11: note: in expansion of macro 'UI_DISPLAY_RS_PIN'

     WRITE(UI_DISPLAY_RS_PIN, rs);

           ^

uiconfig.h:231:31: error: pasting "/* PINK.1, 88, D_RS*/" and "_PORT" does not give a valid preprocessing token

 #define UI_DISPLAY_RS_PIN  8  // PINK.1, 88, D_RS

                               ^

sketch\HAL.h:170:114: note: in definition of macro '_WRITE'

 #define  _WRITE(port, v)   do { if (v) {DIO ##  port ## _PORT -> PIO_SODR = DIO ## port ## _PIN; } else {DIO ##  port ## _PORT->PIO_CODR = DIO ## port ## _PIN; }; } while (0)

                                                                                                                  ^

sketch\ui.cpp:743:5: note: in expansion of macro 'WRITE'

     WRITE(UI_DISPLAY_RS_PIN, rs);

[... 2029 lines cut ...]

sketch\HAL.h:170:147: note: in definition of macro '_WRITE'

 #define  _WRITE(port, v)   do { if (v) {DIO ##  port ## _PORT -> PIO_SODR = DIO ## port ## _PIN; } else {DIO ##  port ## _PORT->PIO_CODR = DIO ## port ## _PIN; }; } while (0)

                                                                                                                                                   ^

sketch\ui.cpp:810:5: note: in expansion of macro 'WRITE'

     WRITE(UI_DISPLAY_ENABLE_PIN, LOW);

     ^

sketch\ui.cpp:810:11: note: in expansion of macro 'UI_DISPLAY_ENABLE_PIN'

     WRITE(UI_DISPLAY_ENABLE_PIN, LOW);

           ^

exit status 1
pasting "/* PINK.1, 88, D_RS*/" and "_PORT" does not give a valid preprocessing token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Nibbels, 2018-09-05:
Ja, da muss der Kommentar hinter dem Define weg. Ich glaube, die Arduino 1.8.6 ist das noch etwas strikter geworden. Hatte auch solche Fehler drin, die früher aus irgendwelchen Gründen nicht aufgefallen sind?

#define UI_DISPLAY_RS_PIN 8 // PINK.1, 88, D_RS

-->

// PINK.1, 88, D_RS
#define UI_DISPLAY_RS_PIN 8

SplitX1, 2018-09-05:
Thanks worked for every error that popped up i hope they update the download file here so noone else has to experience this bug :D

Vielen Dank für jeden Fehler, der auftauchte. Ich hoffe, dass sie die Download-Datei hier aktualisieren, damit niemand diesen Bug erleben muss: D

closed 2018-09-05 by SplitX1