repetier/RepetierFirmware #823: [Feature request] Printing time

luxarts, 2018-09-11:
Hello!

It's very useful to see the current print time in the main screen. As the printed time is stored all you have to do it's at the begining of the print store that value in a new variable and show the difference between both values (current time - start time).
I added it very roughly because I'm modified the firmare to adapt my hardware but was easy to archieve and doesn't affect the calculation time, print speed or anything similar (at least in 32 bit cpu).

Regards,
Lucas

repetier, 2018-09-11:
Where do you get the time from in firmware? E.g. if printing from host/server firmware does not know how much is coming. Therefore we do these computation there and send them to the firmware during the print.

luxarts, 2018-09-11:
Using `HAL::timeInMilliseconds()`. I know about the host capabilities but in my case I don't want to use any host as I'm developing the interface in a web server (similar to octoprint but using an ESP12F module)

repetier, 2018-09-12:
So that will then compute the real printing time? I mean timeInMilliseconds is only time since printer start and does not say how long the print will take.

luxarts, 2018-09-13:
I think you misunderstood me. What I want to watch is the current print time, not the time that it will take. 
So when you start to print the current milliseconds will be stored in a variable, then every screen refresh you will display the actual milliseconds minus the milliseconds that you stored before.

And Arduino like aproach would be:
```
unsigned long startTime = 0;
void startPrint(){
  startTime = millis();
}
void refreshScreen(){
  displayPrint(millis() - startTime);
}
```