repetier/RepetierFirmware #824: non-smooth movement after print around 30 mintues

WindoC, 2018-09-23:
I found a problem. I can't solve it after I have been working for a few days. Don't know where or what cause this issue.

Repetier-server + Repetier-fw 1.0.2 + ramps-fd v2

There is no such problem when I just booted up, but after printing for about 30 minutes, the fast move will become non-smooth (starting at 1:05 in the video)
https://youtu.be/YTR5p8DttvU?t=1m5s
(Seem there are a short delay between each line of gcode)

Same gcode execule after reboot will be no problem (see 4:07 in the video)
https://youtu.be/YTR5p8DttvU?t=4m7s

I have also tried using 0.92.9 and 1.0.1dev and the results are the same.

[testmove.gcode.txt](https://github.com/repetier/Repetier-Firmware/files/2408946/testmove.gcode.txt)

repetier, 2018-09-24:
Surry, but I do not recodnice non smooth moves in the video. I see that at 1:05 the speed increases and sometimes I have the feeling starting a move might be delayed, but moves it self look good in the video.

You should enable logging in server and see if oyu get communication problems when this happens. That can causes small pauses also normally it is a second and nothing in the video looked like a second.

WindoC, 2018-09-24:
It can find the different by count the running time by stopwatch.

I have tried few more test and found the problem not related to print time but for the retraction speed.

If the retraction speed set to 70 mm/s (95step/mm) the non-smooth problem will be worse then 20mm/s.

If turn off retraction. I cannot feel any non-smooth problem.

This is the movement test gcode:
[testbed_180ms.gcode.txt](https://github.com/repetier/Repetier-Firmware/files/2412225/testbed_180ms.gcode.txt)

Log for print test turn off retraction.
[Ripple_test-noretraction.log](https://github.com/repetier/Repetier-Firmware/files/2412241/Ripple_test-noretraction.log)

Log for movement test after printed without retraction. (time: 21s)
[testbed_180ms-smooth-21s.log](https://github.com/repetier/Repetier-Firmware/files/2412237/testbed_180ms-smooth-21s.log)

Log for print test with retraction speed 20mm/s.
[Ripple_test-withretraction_speed20mm.log](https://github.com/repetier/Repetier-Firmware/files/2412266/Ripple_test-withretraction_speed20mm.log)

Log for movement test after printed with retraction 20mm/s (time: 21s)
[testbed_180ms-retraction_speed20mm-21s.log](https://github.com/repetier/Repetier-Firmware/files/2412269/testbed_180ms-retraction_speed20mm-21s.log)

Log for print test with retraction speed 70mm/s.
[Ripple_test-withretraction_speed70mm.log](https://github.com/repetier/Repetier-Firmware/files/2412302/Ripple_test-withretraction_speed70mm.log)

Log for movement test after printed with retraction 70mm/s (time: 26s)
[testbed_180ms-retraction_speed70mm-26s.log](https://github.com/repetier/Repetier-Firmware/files/2412303/testbed_180ms-retraction_speed70mm-26s.log)

repetier, 2018-09-25:
Ok, I think I understand what you mean and I thought it might be the jerk calculation changing the speed. Analysing this I saw that it uses speedE and in calculateMove the speed was not set to 0 if not used. So what I guess is that you had some moves with retracts e.g. at 70mm/s and that required the jerk or join speed to get reduced more then needed.

I have uploaded it as dev version in config tool and github. Please test if the dev version now does not show the problem. But from what I think might happen this should fix the problem. With increasing retract speed this also increases the stored e speed and thus reduces speed at joint even more so this would also fit your description.

WindoC, 2018-09-25:
Yes. Confirm it fixed. Thank for your support!

[image]

closed 2018-09-25 by WindoC

Nibbels, 2018-09-25:
@WindoC Thank you for your precise research on this! I cant wait to flash the update to my printer.

repetier, 2018-09-30:
Ok, is now also included in official master branch as 1.0.3. Thanks for finding the problem.