repetier/RepetierFirmware #825: [Feature request] Stop heating (interrupt busy: heating)

luxarts, 2018-10-11:
Hello!
I noticed that when heating the following commands are not executed until reach the target temp. I watched post here and in the Repetier forum (reprap forums also) looking for a solution but I didn't found it. As you say in a [post](https://forum.repetier.com/discussion/comment/17869/#Comment_17869) it's not possible because the heating commands blocks completely the execution of the buffer and I think I've got a solution:

Every command received by the serial port (any, not readed from SD card) first check if it's an "emergency command" like M108 in Marlin firmware. If it is then break the loop, if not, add to the queue (the buffer).

In code an example would be:
```C++
uint8_t emergencyCommand = 0;

void serialEvent(void){
  while(Serial.available()){
    //Store the byte
    char inByte = (char)Serial.read();
    //Add it to a buffer
    inputBuffer += inByte;
    //If the byte received is new line
    if(inChar == '\n'){
      //If the command received is M108 (emergencyCommand)
      if(inputBuffer[0] == 'M' &&
           inputBuffer[1] == '1' &&
           inputBuffer[2] == '0' &&
           inputBuffer[3] == '8' &&){
        emergencyCommand = 1;
      }
      else{
        addCmdToBuffer(inputBuffer);
      }
    }
  }
}
//Blocking loop
void heatingLoop(void){
  while(currentTemp != targetTemp){
    //Check if the emergencyCommand was received
    if(emergencyCommand == 1){
      emergencyCommand = 0;
      //Break the while
      break;
    }
  }
}
```

I hope it helps!

Regards,
Lucas

luxarts, 2018-10-11:
Also I forgot to mention that after the emergencyCommand is detected the command needs to be cleaned

luxarts, 2018-10-11:
I implemented it in my firmware (custom version of Repetier) adding the following snippet in the Commands.cpp file inside `void Commands::waitUntilEndOfAllMoves()`:

```C++
void Commands::waitUntilEndOfAllMoves() {
    uint8_t inputBuffer[4];
    uint8_t cmdIndex=0;

#ifdef DEBUG_PRINT
    debugWaitLoop = 8;
#endif
    while(PrintLine::hasLines()) {
        //GCode::readFromSerial();
        if(HAL::serialByteAvailable()){             //Check if byte is received
            uint8_t inByte = HAL::serialReadByte(); //Store the byte
            if(cmdIndex == 0){                      //If its the first char
                if(inByte == 'M' || inByte == 'm'){ //If its M or m
                    inputBuffer[0] = inByte;        //Store the first char
                    cmdIndex++;                     //Next char
                }
            }
            else if(cmdIndex == 1){                 //If its the second char
                if(inByte == '1'){                  //If its 1
                    inputBuffer[1] = inByte;        //Store the second char
                    cmdIndex++;                     //Next char
                }
                else{                               //If isn't a 1 (eg M2)
                    cmdIndex=0;                     //Reset the index
                }
            }
            else if(cmdIndex == 2){                 //If its the third char
                if(inByte == '0'){                  //If its 0
                    inputBuffer[2] = inByte;        //Store the third char
                    cmdIndex++;                     //Next char
                }
                else{                               //If isn't a 0 (ex M11)
                    cmdIndex=0;                     //Reset the index
                }
            }
            else if(cmdIndex == 3){                 //If its the second char
                if(inByte == '8'){                  //If its 8
                    break;                          //Break the while loop
                }
                else{                               //If isn't a 1 (ex M100)
                    cmdIndex=0;                     //Reset the index
                }
            }
        }
        checkForPeriodicalActions(false);
        GCode::keepAlive(Processing);
        UI_MEDIUM;
    }
}
```
Added comments to understand what I did.
I think it's better if you add it to the firmware directly instead of make a pull request because my firmware version it's very modified.

repetier, 2018-10-11:
Nice idea and a good.

I see 2 things I'd need to change in host/server for it to work: 
1. The host needs to send the command M108. Normally host will wait for ok and then send command. That must be changed in our host/server. If we reduce used buffer my the size of an emergency command that should be possible. You should know that we fill the input buffer of firmware with commands for faster communication.
2. Last point is if communication is in binary format this would also not help. But that would get difficult. I think I should change it to ascii in host/server for this special case.
At the same spot it would also be possible to catch M112 as well for faster reset detection.

What I do not understand is this part:
```void serialEvent(void){
  while(Serial.available()){
    //Store the byte
    char inByte = (char)Serial.read();
    //Add it to a buffer
    inputBuffer += inByte;
    //If the byte received is new line
    if(inChar == '\n'){
      //If the command received is M108 (emergencyCommand)
      if(inputBuffer[0] == 'M' &&
           inputBuffer[1] == '1' &&
           inputBuffer[2] == '0' &&
           inputBuffer[3] == '8' &&){
        emergencyCommand = 1;
      }
      else{
        addCmdToBuffer(inputBuffer);
      }
    }
  }
}```

I understand serialEvent is called on a byte coming from serial. But Serial.read returns the char so why do you add the char to a pointer. And where does inputBuffer and addCmdToBuffer come from? Is that one of your modifications? Compiler did at least not compile the code, so parts seem missing.

luxarts, 2018-10-11:
The code above it's an example. The code in the [next comment](https://github.com/repetier/Repetier-Firmware/issues/825#issuecomment-428790360) is the one that works. The only modification that I did was replace the break for a return and before it return call Printer::setAutoreportTemp(oldReport); to stop reports.

For the things that you comment:
1. You can replace the function in Repetier-Host/Server of the Stop button to send the M108 command to stop the print. I know you fill the buffer and that's why I read the bytes directly in the while loop. Of course you have to add the M108 command in Commands::processMCode(Gcode *com).
2. Communication in binary shouldn't be a problem. The concept is the same. Send the bytes and read them inside the while loop and not as a normal command in the input buffer.

The M108 command detection algorithm (the one that I wrote in the comment before) has to be implemented before the adding the command to the queue in the buffer. Maybe in the AVR architecture it will slow a bit the execution but in DUE it's insignificant.

Nibbels, 2018-10-11:
I would as well apreciate a solution for a ultra fast M112 detection!
Some friends of mine had this discussion some weeks ago but the problem is still unsolved.

(In the mensioned discussion case it might as well have been a problem with simplify3d.)

luxarts, 2018-10-11:
You can implement the same algorithm with M112. The only change will be when the last char is detected (in M112 will be the 2) then call the software reset function.

Nibbels, 2018-10-11:
Thanks, but well ... the thing is: I know the function GCode::readFromSerial() as the "owner" of all incomming serial bytes. This function feeds GCode::parseBinary and GCode::parseAscii.

Therefore in my mind all bytes are kind of private to these functions. If I start tempering with the serial buffer from another functions view I will risk implementing stupid bugs for sure.
My problem is, that I have no debugger, I always have to test code "live" on my printer. That gives me quite a hard time when it comes to overflows or tricky loop/index bugs.
And it is really annoying having friends telling me that "something does not work well,".

The source of all problems is that we want to have some GCodes to go into some priority line.
If we print from SDCard there should always be one free slot for incoming USB-Commands. But If I print from USB there probably is none. That is a problem if friends print from Simplify3D or another host. 
So what I want to say is, that your solution might work well for your Setting but other printer owners have other weird ideas how to configurate their machines or what host software to use.

Well.... The mensioned functions have some error logic. And the string "M112" might be just a komment! And what about "m112" ...
So parsing the buffer into real Gcodes using Fletchers checksum correction is mandatory for me.
[If I want to produce random faulty serial data I just have to push the extruders steps/mm to like 2560 and maybe activate advance and I can see the faulty data within my serial console. -> errorcorrection is mandatory]
The surrounding code is very hard to read. I was able to understand a lot, but when it comes to buffers and parsing I personally know to less about the background strategies.

I think ghost-reading the buffer from time to time without adding Gcodes to the gcode buffer is a great idea,
But I cannot trust myself to do this in a clean way.

repetier, 2018-10-12:
You have some good points here. M112/M108 could be a com error or part of M117. So to have a reliable solution you need to parse the command. Then you can be sure that comments are stripped and checksum matches. It would also work for binary and ascii and it will not depend on the format used by host, e.g. if send with or without line numbers.

I looked into Marlin solution and it is quite similar. In their serial implementation they check for a sequence like in above example only a bit more complex. But I guess they can also trigger wrongly as it contains no checksum/M117 test.

So how could it be solved otherwise in a clean way. One solution would be to increase preparsed commands but that requires lots of memory. Each gcode takes nearly 100byte and 127 byte binary packed commands can contain at worst 31 gcodes. So the only really working solution is to parse incoming data, only a bit different. We need a preparser that collects data in a special string and always if a end is reached we do a precheck for emergency. That would even allow out of order execution for special command and can be much more general. I think for example that it could even allow babystepping being executed remotely much closer. That would be a great gain in possibilities. I only have to think about how to achieve this in a secure way espcially knowing that we can handle multiple datasources at once, so solution must cope also with 2 serial connections in parallel.

luxarts, 2018-10-13:
What if you make a command (for example M15) to act instantly (not go into the buffer) where the param is the command that you want to execute?

E.g.: 
`M15 M112` -> Reset instantly the printer
`M15 M117 Hello!` -> Display the message "Hello!" in the display
`M15 G28` -> Home instantly
`M15 M290 Z0.1` -> Raise the Z axis 0.1mm (z babystep)

I think you get the point and can see the useful that would be a command like this.

About the detection of the command:
I know you want to implement it in the binary way and with the checksum because you worked hard to do it (and we appreciate that) but let's start with the basic thing: Detect the pattern of this especial command when it arrives to the input serial buffer (the serial buffer, not the queue buffer). In fact that's not difficult as you can see the example in my [first message](https://github.com/repetier/Repetier-Firmware/issues/825#issue-368910337). The only thing you have to do after detecting the "important command" (in the above example M15) is get the param part and then parse it like a normal command that is in the first place of the queue.

I'll download the latest version, make this and send a pull request when I have a break time.

Regards,
Lucas

Nibbels, 2018-10-29:
https://github.com/repetier/Repetier-Firmware/commit/8faf45c37cf0c0c630fd18cf71b6f2716cf86e13