repetier/RepetierFirmware #826: Z Probe not homing

aliaboshady123, 2018-10-13:
Hello everyone,
My printer used 3 min endstops fine. And I wanted to upgrade to make it auto level, so I added a Z Probe instead of the Z min endstop. And when I tried homing the first time, the X and Y homed fine, but the Z axis started moving up and down randomly while squeaking. I restarted and tried again, now it homes X and Y and then outputs "X:0.00 Y:0.00 Z:0.00 E:0.0000 " and the Z axis doesn't move at all.
I wrote m119 to see if the Z Probe isn't working, but it's working fine and outputting signals at it should.

My Config:

#define MIN_HARDWARE_ENDSTOP_Z false
//
//
//
#define Z_PROBE_Z_OFFSET 2
#define Z_PROBE_Z_OFFSET_MODE 0
#define UI_BED_COATING 1
#define FEATURE_Z_PROBE 1
#define EXTRUDER_IS_Z_PROBE 0
#define Z_PROBE_DISABLE_HEATERS 0
#define Z_PROBE_BED_DISTANCE 10
#define Z_PROBE_PIN ORIG_Z_MIN_PIN
#define Z_PROBE_PULLUP 1
#define Z_PROBE_ON_HIGH 1
#define Z_PROBE_X_OFFSET 27
#define Z_PROBE_Y_OFFSET 15
#define Z_PROBE_WAIT_BEFORE_TEST 0
#define Z_PROBE_SPEED 2
#define Z_PROBE_XY_SPEED 150
#define Z_PROBE_SWITCHING_DISTANCE 4
#define Z_PROBE_REPETITIONS 1
#define Z_PROBE_HEIGHT 2
#define Z_PROBE_DELAY 0
#define Z_PROBE_START_SCRIPT ""
#define Z_PROBE_FINISHED_SCRIPT ""
#define Z_PROBE_RUN_AFTER_EVERY_PROBE ""
#define Z_PROBE_REQUIRES_HEATING 0
#define Z_PROBE_MIN_TEMPERATURE 150
#define FEATURE_AUTOLEVEL 1
#define FEATURE_SOFTWARE_LEVELING 0
#define Z_PROBE_X1 30
#define Z_PROBE_Y1 20
#define Z_PROBE_X2 190
#define Z_PROBE_Y2 20
#define Z_PROBE_X3 20
#define Z_PROBE_Y3 160
#define BED_LEVELING_METHOD 1
#define BED_CORRECTION_METHOD 0
#define BED_LEVELING_GRID_SIZE 5
#define BED_LEVELING_REPETITIONS 5
#define BED_MOTOR_1_X 0
#define BED_MOTOR_1_Y 0
#define BED_MOTOR_2_X 200
#define BED_MOTOR_2_Y 0
#define BED_MOTOR_3_X 100
#define BED_MOTOR_3_Y 180
#define BENDING_CORRECTION_A 0
#define BENDING_CORRECTION_B 0
#define BENDING_CORRECTION_C 0
#define FEATURE_AXISCOMP 0
#define AXISCOMP_TANXY 0
#define AXISCOMP_TANYZ 0
#define AXISCOMP_TANXZ 0

repetier, 2018-10-14:
Check your homing settings. You need z in endstop to be z probe pin. Select one o fthe preheat variants with preheat temperature 0. There you can set homing position so it is in the allowed area and not at xy = 0 where you can not deploy probe offset. You might also want to raise Z a bit before homing in case you start at z=0. Depends on when probe triggers.

aliaboshady123, 2018-10-14:
Thank you for replying,
I don't think I understood everything you're saying, but I tried some things from which you mentioned.
#define Z_PROBE_PIN ORIG_Z_MIN_PIN
//
//
#define HEATED_BED_PREHEAT_TEMP 0

I also turned the heatbed on about 60 degrees, but it didn't work.
Still not homing Z axis.