repetier/RepetierFirmware #831: MKS MINI12864 LCD

gerleimarci, 2018-11-09:
What I need to do to get the MKS MINI12864 LCD working with a MKS GEN L? I set `FEATURE_CONTROLLER` to 1 in Configuration.h and `UI_DISPLAY_TYPE` to 5. I have commented out `#define U8GLIB_ST7920` and uncommented `#define U8GLIB_MINI12864_2X_SW_SPI`.

What else I need to change? The display works perfectly under Marlin.

repetier, 2018-11-10:
// For software spi assign these definitions
// SCK Pin:  UI_DISPLAY_D4_PIN
// Mosi Pin: UI_DISPLAY_ENABLE_PIN
// CD Pin:   UI_DISPLAY_RS_PIN

These pin numbers must match as well for software spi.

I think theat shoul dbe all for the display part. Key part must of course also use right pin numbers.

gerleimarci, 2018-11-10:
I used this for reference: [https://reprap.org/wiki/MKS_MINI_12864](url)
```
#define SCK_PIN       52
#define MOSI_PIN      51
#define DOGLCD_CS     25
```
In the Configuration.h:
```
#define UI_DISPLAY_D4_PIN	52
#define UI_DISPLAY_ENABLE_PIN	 51
#define UI_DISPLAY_RS_PIN	25
```
But no success.