repetier/RepetierFirmware #840: stepperWait compile error in HAL.cpp

TwoPony, 2019-01-03:
Just pulled a zipfile of the master of this project.
Also just downloaded a new pull of the Arduino IDE (3 jan 2019 )
In Arduino IDE, the compile fails with an error report:
*** begin of error message ******
Arduino: 1.8.8 (Mac OS X), Board: "Arduino Mega ADK"

/var/folders/1j/lftpdt857czg9hng1q4c5h_w0000gp/T//ccvu1BqJ.ltrans4.ltrans.o: In function `__vector_17':
<artificial>:(.text+0x1c0): undefined reference to `stepperWait'
<artificial>:(.text+0x1c8): undefined reference to `stepperWait'
<artificial>:(.text+0x1d0): undefined reference to `stepperWait'
<artificial>:(.text+0x1de): undefined reference to `stepperWait'
<artificial>:(.text+0x1e2): undefined reference to `stepperWait'
/var/folders/1j/lftpdt857czg9hng1q4c5h_w0000gp/T//ccvu1BqJ.ltrans4.ltrans.o:<artificial>:(.text+0x1e8): more undefined references to `stepperWait' follow
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Mega ADK.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
*** end of error message. ********************

By adding a 'volatile' keyword to the declaration of the stepperWait variable on line 678, I'm able to get rid of the error messages.  Is this a good thing?

repetier, 2019-01-04:
Yes volatile is ok if it helps convincing the compiler to not optimize it away. That is why it was makred ((used)) but looks like compiler is not convinced enough by that any more. It can not analyse it correctly due to the variable mainly being changed in assembler code.

repetier, 2019-01-04:
On windows using 1.8.8 it compiled fine. Have added the volatile for safety in latest dev version.

closed 2019-01-04 by repetier