repetier/RepetierFirmware #842: Not correct to reference CoreXYZ in code?

kyrreaa, 2019-01-07:
From all I find in the code and config tool, it is CoreXY_z or CoreXY+Z as it does not appear to support proper XYZ operation with 3 or 4 motors.

I've just built a fully 3-axis CoreXYZ and was really hoping I could just set it up with Repetier.
Seems like I have some work to do :)

Can someone clarify this?
Apparantly Ryan Carlyle did some work on this a while ago, but I have not found any updated info on it.

repetier, 2019-01-08:
You mean mirroring the motors? Might in deed miss here in that case. Core uses different parts to set signals. You can use FAST_COREXY which uses nonlinear solution. Not 100% sure but it use different code to set motors.

You can try to switch directly to V2 if you have the right components especially a due based board. There it is supported.

kyrreaa, 2019-01-08:
Really?
I can't even get dev2 branch to load properly in the Studio Code editor.
(Hate this editor btw, seems designed by someone who learned esthetic ui design and ignored all indicators on what indicates a button and gives usability.)
It also installed a tonn of crap "recommended by the project" that then is incompatible with itself.

**What is the proper way to open this project?**
(Did you ever look at EmBitz by the way?  Quite capable editor as well, free, although not open source.)

kyrreaa, 2019-01-08:
Looking at the code it still appears to me to really just be working with H/XY style gantry machines.
I cannot find enough motor outputs from the transforms.

Motor naming: (As seen from abowe)
Back
D   C
A   B
Front

Motor rotation is seen from shaft side of motor; face on.
Contributions from Motors ABCD on motion XYZ: (+N denotes positive axis motion contribution from clockwise rotation of motor N.)
X = +A +B -C -D
Y = +A -B -C +D
Z = +A -B +C -D

The ABCD data would then be:
A =  X + Y + Z
B =  X - Y - Z
C = -X - Y + Z
D = -X + Y - Z

It thus takes cartesian input and generates steps on 4 motors. None of which can just be paralleled...

kyrreaa, 2019-01-08:
I think I figured out the problem with the dev2 branch. It sais in the .md to install Arduino extension for Visual Studio Code, but it is incompatible with platform.io and as soon as I uninstalled it stuff started working.

On a side note, the way motors are set up in config for all boards in repetier has not changed I see, so they are still called x,y and z.
a, b, c, d, e, f would be more natural since it makes sure there is no confusion about motor level control and axis level. For cartesian one can then assign the motors to axis, or for deltas and other setups; assign motors to transform outputs.

In order to control motor D now, I guess a quick hack is to add transform to a->x, b->y, c->z and d->a, and then expand the mix by defining #COREXYZ_A_X, #COREXYZ_A_Y and #COREXYZ_A_Z and adding the extra printer type to handle the real xyz case.
It seems doable :)

repetier, 2019-01-09:
I think you miss one important part - motor drivers are generic. So if you need 2 motors for x see
https://docfirmwarev2.repetier.com/config/motors
STEPPER_MIRROR2(name, motor1, motor2, minEndstop, maxEndstop) 
is what you then use for x and as input you have 2 motor definitions for x1 and x2 that get merged to x. Same works for 3 or 4 motors. And because it is generic you can do this with every motor input. So there is no special handling required in the handling code for mirroring.

Arduino IDE is what I started with and it was a pain. Will search that hint and remove it from docs. That was a expired solution.

kyrreaa, 2019-01-09:
I need 4 motors for x, 4 for y and 4 for z. But it is the same 4.
Dependibg on the motion vector each of the 4 will run in a specific direction and a potentially different number of steps.
The actual mechanics is not important for the drive, but the relative mixes and polaritys are.

repetier, 2019-01-09:
How can 4 motors move 3 axes and none is a duplicate? That does not look like well defined math. Normally you have 3 targets defined my 3 variables. That sounds like a very experimental thing you have here.

There must be a combination that gets 4th motor position from 3 other one.

Anyhow, guess no firmware has such a combination implemented. What you can do is make your own printer system like the 3 existing. Just one that has 4 dependent axis. After all V2 supports up to 7 independent axes. You still have X,Y,Z as input but also use A-Axis for the 4th motor. You can more or less 
use the cartesian solution and just modify the transformation

```
void PrinterType::transform(float pos[NUM_AXES], int32_t motor[NUM_AXES]) {
    FOR_ALL_AXES(i) {
        motor[i] = lroundf(pos[i] * Motion1::resolution[i]);
    }
}
```

to use a new mapping to 3 motors ignoring A input value. That should do the trick. That is the only place where V2 converts coordinates into motor positions.

kyrreaa, 2019-01-09:
As you can see, it uses a lifted core-xy principle. This is Core-XY expanded to a full Core-XYZ by direct extension. There is a way to do it with 3 motors, but it does not cancel out torque in frame from motors, or it moves one motor, or it needs a even more complicated belt path.
I plan to use a kevlar cored 2mm non-slip coated driveline for this, similar to what is used in process industry to make beltlike conveyors. Very rugged and wear resistant.

I have to thank Ryan Carlyle for the initial concept for this as he did the work together with a google group to expand the Core-XY system to 3 dimensions.
The benefit is for example a stationary bed and a rigid frame that can carry a high weight, or a light frame for high speed.

I use 12mm steel rods for Z axis and 12mm carbon fiber tubes for X and Y.

The usable print area is very large and one can have an array of "tools" on the upper front wall to change from without losing much if any area since no rod is crossing this point without being the actual gantry.
Alternatively the back edge can be used for attaching things to back of carriage as well, for even higher degrees of expandability.

The belt system cancels out all binding forces due to the synchronous and differential driving.

In this video I only drive the two back motors (C and D):
https://youtu.be/WhulIxWNhXw

In this video I drive B and C:
https://youtu.be/amu9ObqnguI

I still claim that this is the only "true" CoreXYZ, and all others have been CoreXY+Z, CoreXZ+Y and so forth...

Anyways; Fun times are ahead...

repetier, 2019-01-09:
AL1Motor and AL2Motor are 2 extra motors I have for my bed autoleveling to physically adjust bed.
You should just have define AMotor as 4th motor. You do not have these 2 auto leveling motors.

```
motor[X_AXIS] = lroundl((COREXYZ_X_X * pos[X_AXIS] + COREXYZ_X_Y * pos[Y_AXIS] + COREXYZ_X_Z * pos[Z_AXIS]) * Motion1::resolution[X_AXIS]);
    motor[Y_AXIS] = lroundl((COREXYZ_Y_X * pos[X_AXIS] + COREXYZ_Y_Y * pos[Y_AXIS] + COREXYZ_Y_Z * pos[Z_AXIS]) * Motion1::resolution[Y_AXIS]);	    motor[Y_AXIS] = lroundl((COREXYZ_Y_X * pos[X_AXIS] + COREXYZ_Y_Y * pos[Y_AXIS] + COREXYZ_Y_Z * pos[Z_AXIS]) * Motion1::resolution[Y_AXIS]);
    motor[Z_AXIS] = lroundl((COREXYZ_Z_X * pos[X_AXIS] + COREXYZ_Z_Y * pos[Y_AXIS] + COREXYZ_Z_Z * pos[Z_AXIS]) * Motion1::resolution[Z_AXIS]);	    motor[Z_AXIS] = lroundl((COREXYZ_Z_X * pos[X_AXIS] + COREXYZ_Z_Y * pos[Y_AXIS] + COREXYZ_Z_Z * pos[Z_AXIS]) * Motion1::resolution[Z_AXIS]);
    motor[A_AXIS] = lroundl((COREXYZ_A_X * pos[X_AXIS] + COREXYZ_A_Y * pos[Y_AXIS] + COREXYZ_A_Z * pos[Z_AXIS]) * Motion1::resolution[A_AXIS]);
    for (fast8_t i = E_AXIS; i < NUM_AXES; i++) {	    for (fast8_t i = E_AXIS; i < NUM_AXES; i++) {
        motor[i] = lroundl(pos[i] * Motion1::resolution[i]);	        motor[i] = lroundl(pos[i] * Motion1::resolution[i]);
    }	    }
```

The for loop should only go < A_AXIS or you overwrite motor[A_AXIS] with 0!

Apart of that it looks good. Hope the printer is stiff enough to not bend with all the belts/wires. Do you have Belts and rotate them around one corner?

kyrreaa, 2019-01-09:
Oh! And to answer your question: Yes there are lots of illegal motions with 4 motors like this.
The design is over-constrained to avoid compex split-belt paths.

kyrreaa, 2019-01-09:
Ahh, ok. A bit of a misunderstanding by me there.

And yes, due to the design you can have incredible belt tension.
The belt is carried around on on v-groove bearings and being round it allows changing directions with no twisting.
[image]

For simplicity I am using a friction-amplifier on the motor consisting of just an extra set of bearings.
[image]

This is done to make the printer buildable by anyone without extra tools.
One can order the profiles precut (they are 3030 not 2020 so beefier) and this size machine can use the rods exactly as they come at 500mm lengths.
I use a ziptie-gun to tighten stuff and a bag of screws and t-nuts.
The only magical parts are the aluminium corner brackets which are also easily available and allow a stiff frame without wasting any space in the frame for corner braces.

Drive is done via high density thick walled silicone tubing pressed onto motor shaft. Maybe not ideal, (we'll see), but good enough for testing.

kyrreaa, 2019-01-09:
Yes, that is my concern also. But if we don't build it, we won't know...

kyrreaa, 2019-01-09:
Any idea how to debug with Studio Code + PlatformIO?
It doesn't seem to allow mt to use any of my debuggers. I got all of Atmels stuff.
I thought the simplest to use would be the jtagice3 as that has correct header already provided on the DUE.
However, PlatformIO keeps nagging me about an account which I will not do for a local debug session...

kyrreaa, 2019-01-10:
Debug requires account but noting more apparantly now.
Anyways, it is faulting and debugging is usefull...
Do you need a debugger? I can donate a new Atmel jtagice3. At the moment it is not behaving well with platformio though. Apparantly I need to fiddle with something in settings or params.

repetier, 2019-01-10:
I have a jtagice but have currently no problems on my side.

kyrreaa, 2019-01-10:
Is that the ice3?
Tiny one with the micro ribbon cable to 10 pin 0.05" (1.27mm) pitch?
My Arduino Due board has header for that but all I get are problems.

I switched to a STlink and managed to see that it halts due to a fault handler but it was not stable.
Neither would program.

kyrreaa, 2019-01-10:
Ahh, yes mine works perfectly in Atmel Studio too.
I still have a Atmel JTAGICE3 you can have. You'd have to message me your address somehow.
I rely heavily on these debuggers for all my work as they cut down fault finding to seconds (when they work).

repetier, 2019-01-10:
Ok, now I have managed to test it with my JLink debugger, after being fooled to think it is Atmel-Ice and needing to kill 40 pio processes to get it back working. It is really nice if you can simple stop anywhere and I even see the variables now. Don't remember having seen them when I debugged in Atmel-IDE.

What I also learned is to disable watchdog when debugging or every break causes a reset on continue.

Only bad think is, that it is only free for 30 days. Then you need a paid account to debug. Ok for me as I need it more often but bad for part time developers.

repetier, 2019-01-10:
Just downloaded it but it only contained an exe file, so I assume there is only a windows version. How did you add arduino support? Did you just add the arduino core library? I also see no AVR support when I create a project. So I guess it is nice if it fits your needs but not usable with the wide range of usages our firmware uses. Here PlatformIO is making thinks really simple so far supporting different editors and platforms and frameworks.

Only thing is which I really do not like is needing to pay for open source debug tool integration. On the other hand I understand that you need to earn money somehow if projects make much work.

kyrreaa, 2019-01-10:
Yes, for me windows only is not a big issue as in office environments with integrated calendars and stuff it is almost necessary anyways. Also, Arduino support is non existant so I just copy in the precompiled elfs from Arduino IDE and link with those when debugging.

I think the Arduino plugin to VSCode would be better than PlatformIO though as it does not contain any account links.
We will see what happens in future.

kyrreaa, 2019-01-10:
I just did a test with PlatformIO in VSCode and created a new project for a stm32f4discovery which I took from a sealed bag today. Brand new.
This board has integrated debugger so I know it works.
I added just a serial.begin and a println to setup on a blank Arduino project.
Then I started debug. It failed.
It did manage to load the debugger and connect to the chip. It however failed to detect chip size and from there it was just a mess.

So I opened EmBitz and asked it to create a cmsis library based blank project for the discovery.
Then I just hit debug and it immediately loaded the compiled code onto the discovery and presented me with a debugger view of the code halted in main. Everything worked.
So, something is broken in PlatformIO...

Will dev2 work on something non due?
I assume that with the Arduino framework I should be able to atleast test the code changes for the kinematics?

kyrreaa, 2019-01-10:
By the way... By extending the transform to use A_AXIS I was expecting the non linear (FAST_COREXYZ) to create motion in A as well, but it does not.
"G0 Axx" however does. If I remove the transform for X_AXIS data in the same function it disables the motion of X so I am clearly running and editing the right part of the code.

Is there some other place where motion is triggered/suppressed and that is only looking for x,y,z motion from any such command? To me it looks like the path planner is taking all the transform data from the xyz command even though it is not containing a A but no motion on A...

kyrreaa, 2019-01-11:
https://youtu.be/Ii0lUfQ8pJI
There will have to be some polishing on this before it can ever be mainline, infact I think it may need a lot of reorganizing to be in dev2 hehe.
But... <starwars reference>"**It's working! It's working!**"</starwars reference>!

kyrreaa, 2019-01-11:
So, the question is how is this best done in future?
For a non linear machine, or for a transform based machine given an arbitrary transform, the machine job coordinates are always (usually?) in x,y,z (not for turning?) and there is currently never more than 3 outputs from the transform, and they are connected to x y and z motor.

For this to be properly modular, I would think the transforms which can already take all known axis (like xyz,e1,e2,a,b,c) as input should result in output that would be mapped to motor[0:n]. Since you already support for example stepping x axis on multiple motors there is some of this mapping going on already.
This would allow the same system to do those rotary-table type bots as well with just a different transform mapping to a x and z motor while sending the rotary data to the a axis. (In cnc a is usually the firxt extra axis and many cnc's use a rotary table.) This leaves the y axis unmapped and such a machine could very well have a normal xyz mode as well to change tools.
Anyway, a slight change to the way the input axis end up as output on steppers would help here.

repetier, 2019-01-11:
In V2 everything is handled as nonlinear. The transform function though is the only place where this transformation is needed as it is.

The main problem is that different printer types have special needs like deltas always home to top. So that is what I tr yto solve with the printer types. They handle all the printer specific parts in a .cpp/.h file and based on printer type one of the variants is selected. So writing a new type just means adding these 2 files and the switch to include the right one and you are done. So rotary bot is also just a special version that has it's own homing logic, which is the biggest difference. Rotary table just ha sthe problem of singularity at rotation axis. So it should be forbidden to print in that area or cross it. That is a problem. 

For your solution I might add it simply by testing if the factors for A axis are defined. Homing logic etc is the same so that is lucky coincedence.

repetier, 2019-01-11:
Ok, for next update I modified transform into

```
void PrinterType::transform(float pos[NUM_AXES], int32_t motor[NUM_AXES]) {
    motor[X_AXIS] = lroundl((COREXYZ_X_X * pos[X_AXIS] + COREXYZ_X_Y * pos[Y_AXIS] + COREXYZ_X_Z * pos[Z_AXIS]) * Motion1::resolution[X_AXIS]);
    motor[Y_AXIS] = lroundl((COREXYZ_Y_X * pos[X_AXIS] + COREXYZ_Y_Y * pos[Y_AXIS] + COREXYZ_Y_Z * pos[Z_AXIS]) * Motion1::resolution[Y_AXIS]);
    motor[Z_AXIS] = lroundl((COREXYZ_Z_X * pos[X_AXIS] + COREXYZ_Z_Y * pos[Y_AXIS] + COREXYZ_Z_Z * pos[Z_AXIS]) * Motion1::resolution[Z_AXIS]);
#if defined(COREXYZ_Z_X) && defined(COREXYZ_Z_X) && defined(COREXYZ_Z_X) && NUM_AXES > A_AXIS
    motor[Z_AXIS] = lroundl((COREXYZ_A_X * pos[X_AXIS] + COREXYZ_A_Y * pos[Y_AXIS] + COREXYZ_A_Z * pos[Z_AXIS]) * Motion1::resolution[Z_AXIS]);
    motor[E_AXIS] = lroundl(pos[E_AXIS] * Motion1::resolution[E_AXIS]);
    for (fast8_t i = B_AXIS; i < NUM_AXES; i++) {
        motor[i] = lroundl(pos[i] * Motion1::resolution[i]);
    }
#else
    for (fast8_t i = E_AXIS; i < NUM_AXES; i++) {
        motor[i] = lroundl(pos[i] * Motion1::resolution[i]);
    }
#endif
}
```

that should cover your case as well.

Regarding motors and names we can discuss quite long and different optinions will always be there. Fact is most boards have named drivers slots with names x, y, z and then normally E0, E1, ... and I decided to name the axis motors the same as stored motors. For the mostly used cartesian system this is nice as it is a 1:1 match. For nonlinear systems we have names for coordinates that do not match motor names, yes. That is something you have to describe and the docs contain that I think. For each printer type I write docs on how to define the motors so that is not so hard to understand. If I give them just numbers you would also need a doc describing that M1 is the motor fox x axis or left front pillar on delta. So I think we should stop that discussion here. I do not plan renaming them and we can easily find users liking or not liking this. But physical motor names are so at the end of the process that it does not make a difference for 99% of the code to understand. Only in Motion level 2 it will start mapping to physical motors, all other functions will use the axis names.

kyrreaa, 2020-02-23:
How are things going on the separation of the logical axis of motion input (gcode) and the motors?
I have been looking at dev2 branch and it seems to still need platform.io?
I've finally gotten all my files worked out for the 3d model as I had to replace all the string with proper timing belts. Core-XYZ cannot accept slippage due to the overconstrained nature.

repetier, 2020-02-24:
There are no plans to not use platformio for V2. In combination with Visual Studio Code it is as easy as arduino only 10 times better. Arduino can not handle all the subdirectories correctly.

>  Core-XYZ cannot accept slippage due to the overconstrained nature.

What do you want to say with that? V2 allows any linear combination of x, y and z.

kyrreaa, 2020-02-24:
I was just saying why I could not use my wire-drives on the prototype printer I built.
I have rebuilt it for belts and it has taken me a year...

Platform.io is actually a lot better now than a year ago so this was a joy to use!
Moving my changes to the latest dev2 worked ok, but I seem to get double the steps on first motor.
Trying to enable debug to see if it is calculating too many steps there or just outputting too many pulses but I cannot debug moves as it claims 'PrintLine' not defined in GCode_0_1(). I guess it's not the right flag to enable.

kyrreaa, 2020-02-25:
I feel so stupid... I grabbed a 1.8deg motor by mistake from my pile of motors. The 3 others are 0.9deg/step. No wonder it acted wrong but I could not find any error. There is none.
Video coming soon :)

kyrreaa, 2020-02-25:
How would this be homed when the axis defined are X, Y, Z and A for motors, but homing has to be done on the virtual input axis X, Y and Z. Transformation happens from X, Y and Z into X, Y, Z, A through the PrinterType::transform().
https://youtu.be/Dj_EACiAcdg

repetier, 2020-02-26:
The homing is done in global directions not motor directions, so that is not a big problem. Set homing direction 0 for A so it never homes. Just hope that is no problem for motion. Depending on where your endstops are you can use homing order to do the homings in right order.