repetier/RepetierFirmware #844: Timing of execution

rafaa1988, 2019-01-14:
Does Reprap (repetier) firmware in Arduino execute the sent commands from pc, slicer program, raspberry pi, octorpint, etc.... line by line or does it wait? How many lines or commands does it take until it starts execution? If I want to synchronize the command sent from my pc to the arduino with a command sent from my pc to any other device, or card. So that both commands are executed at the same time, then how to adjust this?

Many thanks for help

repetier, 2019-01-14:
Complicated. Firmware has a active command that it is executing (or is empty) plus it can have several commands that are already send by host software and waiting for execution. To be able to optimize moves, they are cached and executed in background while firmware continues to process following commands. If move cache is full it will wait for a free slot and continue parsing. So as you see in the state of gcode being send you can not be sure when a command will be executed exactly.

So it is important to know what you want to achieve. You can for example send 30 M400 and be sure that as soon as the next command gets send it will be only a few milliseconds until it gets executed. So in host/server you can use @execute after that directly to have both nearly in sync. Now M400 will wait for motions to end - that position will be visible in print. Main reason for the motion cache. So that is only good at positions where it is not visible.

kyrreaa, 2019-01-14:
How about having a way to echo line numbers when they are done executing?
That way host can display the actual printer position instead of displaying animation of what has been sent to printer? Would atleast synch em up better?

repetier, 2019-01-15:
Enabling echo does this. Echo is done by default after execution. But that requires someone to read output and also causes much more communication. But at least it does not require a wait. On the other side if it needs to sync to moves it will not work as moves are executed async.

kyrreaa, 2019-01-15:
Isn't there a big difference between echoing entire lines and just the line numbers?