repetier/RepetierFirmware #846: Error when opening repetier in arduino 1.8.8

WD24, 2019-02-06:
When I open repetier in Arduino 1.8.8 i get the following error at the console window.

"Communication.cpp" contains unrecognised characters. If this code was created with an older version of Arduino, you may need to use Tools -> Fix Encoding & Reload to update the sketch to use UTF-8 encoding. If not, you may need to delete the bad characters to get rid of this warning.

using Fix Encoding & Reload does not remove the message.

badvision, 2019-02-16:
Have you tried Platform.IO?  I've been able to compile on Linux and Windows (via VS Code plugin for Platform.IO) with no issues.

repetier, 2019-02-16:
Nice idea using Platform.IO - I already use it for V2 development, but did not want to break Arduino Compatibility for V1. But maybe there is a way to compile it with both IDEs. Would make developing easier.

Regarding warnings - you can ignore them. I have seen them also I do not know where exactly they come from. Code should be ascii but there seems to by some ISO8859 chars as well in comments or in some translations.

badvision, 2019-02-18:
It's fine to leave the 1.x branch set up for Arduino.  You literally just have to import the project if using VSCode into your workspace from the Platform.IO home tab.   For linux, you do... absolutely nothing.  Just works. :)

repetier, 2019-02-18:
I have updated dev version for PlatformIO already. Found the src_dir option so I can leave the directory name for arduino. So now you open ArduinoAVR/DUE with platformio and are done or you open Repetier.ino in Arduino IDE. Both seem to work in parallel. So that problem is fixed.