repetier/RepetierFirmware #848: PID autotune - .stopDecouple();

Nibbels, 2019-02-11:
https://github.com/repetier/Repetier-Firmware/blob/c09783a0f4f1bc2bedefc8df2c90709172ae74b2/src/ArduinoAVR/Repetier/Extruder.cpp#L2384

Shouldn't we insert a 
	.stopDecouple();
here?
I am researching a case that I had a decoupling error after a PID autotune with possibly preset temperatures.
So my guess is, that I have been within the second phase of decoupling test and then started an PID autotune.

We have this check: https://github.com/repetier/Repetier-Firmware/blob/c09783a0f4f1bc2bedefc8df2c90709172ae74b2/src/ArduinoAVR/Repetier/Extruder.cpp#L128
(And a clone of this line 5 lines down which I think could be removed) https://github.com/repetier/Repetier-Firmware/blob/c09783a0f4f1bc2bedefc8df2c90709172ae74b2/src/ArduinoAVR/Repetier/Extruder.cpp#L133
So my guess is, that if if the decoupled test was started we just delay it, play with temps and then it thinks that our temps went away.
Is that correct?

I will test tomorrow if a 
    extruder[controllerId].tempControl.stopDecouple();
after 
https://github.com/repetier/Repetier-Firmware/blob/c09783a0f4f1bc2bedefc8df2c90709172ae74b2/src/ArduinoAVR/Repetier/Extruder.cpp#L2384
resolves the situation for me. But my trouble is that I dont really know how I produced the bug at first. So reproducing might take some trys.

(I write here because I wanted to check If I had missed some point when cherrypicking your decoupling code. But It looked the same. So If you say that is a clear case I am happy - otherwise I have to search another reason for my bug)

Greetings

Nibbels, 2019-02-12:
Yes, that following commit should have fixed it 99%

https://github.com/Nibbels/Repetier-Firmware/commit/aef3f87894bcb6fe56522cee9726a3cf2494c4f1

The problem was there when a heater was ON prior to PID autotune start. And now the autotune does not trigger the decouple test anymore when tempering with those temperatures without the PID tempmanager knowing.

Greetings

repetier, 2019-02-13:
Yes, I think you are 100% right

```
        if(act->targetTemperatureC < 20.0f) { // heating is off
            output = 0; // off is off, even if damping term wants a heat peak!
            act->stopDecouple();
        }
```

is what we do when we set temperature to off. Since autotune modifies handling we would return in unpredicted state, also set temperature normally would be 0. But adding this would also cover the case that temperature is not 0 when starting autotune.

closed 2019-02-13 by repetier