repetier/RepetierFirmware #851: Backlash calibration on CoreXY

3DMindPL, 2019-02-20:
Hello,

I got the problem with the backlash on my custom made 50x30x25 cm machine. It is impossible to avoid backlash mechanically due to very long belts on CoreXY mechanic. 

It is built on Hiwin and rock-solid construction but still, there is a problem. I find out that there is an option to calibrate it on the firmware. Is there an option to do it with CoreXY? Can you tell me how and what should I do? 

Thanks in advance.
Philip

Jmjcoke, 2019-02-20:
Mercurry GT2 Timing Belt Torsion Spring For 3D Printer 6mm Width Belt Pack
of 10
https://www.amazon.com/dp/B01E913IJK/ref=cm_sw_r_cp_api_i_9SxBCbDYHCQA6

3DMindPL, 2019-02-20:
There is no place where I can put it. It will do collisions with the GT2  pulleys. And I got 10mm belt. 
Even if very tighten, same effect.

repetier, 2019-02-20:
Even long belts would not cause backlash. What they will do is change steps per mm, but that is not backlash. Backlash would be if the pulley is not fastened 100% so on direction changes you need some extra degrees before pulley turn in other direction. 

The backlash implemented works only for cartesian printers as expected. For CoreXY you would need cases that will not happen. Reason is that backlash just inserts extra moves on global xy direction changes. With the interconnection to 2 motors simultaneously this is most likely not solving your problem and even if it does it would be a pain for print quality.

3DMindPL, 2019-02-20:
Maybe it is a wrong diagnosis that it is a backlash effect. But the circles are flat on 2 sides (kinda straight lines in X-axis direction). Is it something that I can correct by firmware?

repetier, 2019-02-20:
Would need to see a printed vs. preview comparison to judge that. There is always the chance that the model is flat at sides and slicer generated it that way. But maybe there is some backlash in y direction until carriage changes directions. In that case you could enable backlash and set y to the value (if x is flat then because y did not move so there you need to compensate). Then see if it gets better.

3DMindPL, 2019-02-20:
It look like this :

http://pl.tinypic.com/view.php?pic=2lw9dly&s=9#.XG2Nx7hCdPY
http://pl.tinypic.com/view.php?pic=svswp3&s=9#.XG2OAbhCdPY

Its pretty round but with two sides flat. But albo you can see gaps in the wall. 
But it is CoreXY mechanic, so how can I enable Y backlash?

repetier, 2019-02-20:
Do not really see anything on the photo and one link does not work.

```
/* If you have a backlash in both z-directions, you can use this. For most printer, the bed will be pushed down by it's
own weight, so this is nearly never needed. */

#define ENABLE_BACKLASH_COMPENSATION 1
#define Z_BACKLASH 0
#define X_BACKLASH 0
#define Y_BACKLASH 0.1
```

should do it.
Will not work with the fast corexy version. Only the normal version using cartesian code will work with backlash feature.

3DMindPL, 2019-02-20:
But like I said - I don't have the cartesian mechanic, just CoreXY. Can I use the code also in CoreXY?

Maybe that photo (and drawing) will be better :
https://imgur.com/a/tQZJp5j

repetier, 2019-02-21:
Yes, that is what I said. Only if you use fast corexy it will not work.

Seeing your second image 0.3mm error is not much and explains why I do not see the effect on the photo. You should try rotating stl 90° and reslice/print and see if the error stays at x max/min or if it now is 90° rotated.  Could also be just a segment being flat at that position. Remember that circles are approximated be line segments so parts are always flat.

ruggb, 2019-03-10:
I am experiencing backlash also on my coreXY. The other day the plastic part holding my idler broke and I may have discovered the cause of the backlash. My bearing was jammed between 2 flat washers. I needed a thin spacer on each side of the bearing. I have not reassembled yet to verify that but you might check that out.