repetier/RepetierFirmware #852: Extruder not extruding

TheSheepKing151, 2019-02-21:
Hello,
I installed the repetier firmware on my tronxy X1. 
Everything works fine besides two things:
1: the extruder stepper doesn't move.
2: while connected via USB the printer resets every 10 seconds

I hope that you have some tips for me
Jonas

repetier, 2019-02-22:
Extruder only moves when hot. Also check eeprom settings if all values make sense, especially steps per mm.
M302 S1
would allow cold extrusion as well.

Does printe ronly reset when connected to a host software? Which? Reconnections might cause a reset. Repetier-Server does this if response is unreadable e.g. because baud rate was wrong.

TheSheepKing151, 2019-02-23:
Ok, that was the problem. The extruder does move when its's hot!

After flashing the firmware again, the resets stopped. But i still can't controll the printer via Repetier Host.
It can connect, but i can't manual controll the printer. Baud rate is right.

repetier, 2019-02-23:
What does log show after 
`M119`
If you get a good answer you know connection is working and it is just wrong configuration or wrong eeprom values.

TheSheepKing151, 2019-02-24:
Repetier Host only says 1 command waiting. In the log it says "No start signal detected - forcing start"

repetier, 2019-02-24:
Then you are not really connected. x commands waiting means not all commands were send and this happens if host does not understand printer. Typical reason is wrong baud rate.

TheSheepKing151, 2019-02-24:
Any other causes? Baud rate is 115200 on both machines.

repetier, 2019-02-24:
What do you see in log if all filters are disabled? Malayan boards need a different RTS setting for the serial port then other boards, but you can currently not set it in host, only in repetier-server.

TheSheepKing151, 2019-02-24:
20:20:44.575 : OpenGL version:4.6.0 NVIDIA 416.34
20:20:44.576 : OpenGL extensions:GL_AMD_multi_draw_indirect GL_AMD_seamless_cubemap_per_texture GL_ARB_arrays_of_arrays GL_ARB_base_instance GL_ARB_bindless_texture GL_ARB_blend_func_extended GL_ARB_buffer_storage GL_ARB_clear_buffer_object GL_ARB_clear_texture GL_ARB_clip_control GL_ARB_color_buffer_float GL_ARB_compatibility GL_ARB_compressed_texture_pixel_storage GL_ARB_conservative_depth GL_ARB_compute_shader GL_ARB_compute_variable_group_size GL_ARB_conditional_render_inverted GL_ARB_copy_buffer GL_ARB_copy_image GL_ARB_cull_distance GL_ARB_debug_output GL_ARB_depth_buffer_float GL_ARB_depth_clamp GL_ARB_depth_texture GL_ARB_derivative_control GL_ARB_direct_state_access GL_ARB_draw_buffers GL_ARB_draw_buffers_blend GL_ARB_draw_indirect GL_ARB_draw_elements_base_vertex GL_ARB_draw_instanced GL_ARB_enhanced_layouts GL_ARB_ES2_compatibility GL_ARB_ES3_compatibility GL_ARB_ES3_1_compatibility GL_ARB_ES3_2_compatibility GL_ARB_explicit_attrib_location GL_ARB_explicit_uniform_location GL_ARB_fragment_coord_conventions GL_ARB_fragment_layer_viewport GL_ARB_fragment_program GL_ARB_fragment_program_shadow GL_ARB_fragment_shader GL_ARB_framebuffer_no_attachments GL_ARB_framebuffer_object GL_ARB_framebuffer_sRGB GL_ARB_geometry_shader4 GL_ARB_get_program_binary GL_ARB_get_texture_sub_image GL_ARB_gl_spirv GL_ARB_gpu_shader5 GL_ARB_gpu_shader_fp64 GL_ARB_gpu_shader_int64 GL_ARB_half_float_pixel GL_ARB_half_float_vertex GL_ARB_imaging GL_ARB_indirect_parameters GL_ARB_instanced_arrays GL_ARB_internalformat_query GL_ARB_internalformat_query2 GL_ARB_invalidate_subdata GL_ARB_map_buffer_alignment GL_ARB_map_buffer_range GL_ARB_multi_bind GL_ARB_multi_draw_indirect GL_ARB_multisample GL_ARB_multitexture GL_ARB_occlusion_query GL_ARB_occlusion_query2 GL_ARB_parallel_shader_compile GL_ARB_pipeline_statistics_query GL_ARB_pixel_buffer_object GL_ARB_point_parameters GL_ARB_point_sprite GL_ARB_polygon_offset_clamp GL_ARB_program_interface_query GL_ARB_provoking_vertex GL_ARB_query_buffer_object GL_ARB_robust_buffer_access_behavior GL_ARB_robustness GL_ARB_sample_shading GL_ARB_sampler_objects GL_ARB_seamless_cube_map GL_ARB_seamless_cubemap_per_texture GL_ARB_separate_shader_objects GL_ARB_shader_atomic_counter_ops GL_ARB_shader_atomic_counters GL_ARB_shader_ballot GL_ARB_shader_bit_encoding GL_ARB_shader_clock GL_ARB_shader_draw_parameters GL_ARB_shader_group_vote GL_ARB_shader_image_load_store GL_ARB_shader_image_size GL_ARB_shader_objects GL_ARB_shader_precision GL_ARB_shader_storage_buffer_object GL_ARB_shader_subroutine GL_ARB_shader_texture_image_samples GL_ARB_shader_texture_lod GL_ARB_shading_language_100 GL_ARB_shading_language_420pack GL_ARB_shading_language_include GL_ARB_shading_language_packing GL_ARB_shadow GL_ARB_sparse_buffer GL_ARB_sparse_texture GL_ARB_spirv_extensions GL_ARB_stencil_texturing GL_ARB_sync GL_ARB_tessellation_shader GL_ARB_texture_barrier GL_ARB_texture_border_clamp GL_ARB_texture_buffer_object GL_ARB_texture_buffer_object_rgb32 GL_ARB_texture_buffer_range GL_ARB_texture_compression GL_ARB_texture_compression_bptc GL_ARB_texture_compression_rgtc GL_ARB_texture_cube_map GL_ARB_texture_cube_map_array GL_ARB_texture_env_add GL_ARB_texture_env_combine GL_ARB_texture_env_crossbar GL_ARB_texture_env_dot3 GL_ARB_texture_filter_anisotropic GL_ARB_texture_float GL_ARB_texture_gather GL_ARB_texture_mirror_clamp_to_edge GL_ARB_texture_mirrored_repeat GL_ARB_texture_multisample GL_ARB_texture_non_power_of_two GL_ARB_texture_query_levels GL_ARB_texture_query_lod GL_ARB_texture_rectangle GL_ARB_texture_rg GL_ARB_texture_rgb10_a2ui GL_ARB_texture_stencil8 GL_ARB_texture_storage GL_ARB_texture_storage_multisample GL_ARB_texture_swizzle GL_ARB_texture_view GL_ARB_timer_query GL_ARB_transform_feedback2 GL_ARB_transform_feedback3 GL_ARB_transform_feedback_instanced GL_ARB_transform_feedback_overflow_query GL_ARB_transpose_matrix GL_ARB_uniform_buffer_object GL_ARB_vertex_array_bgra GL_ARB_vertex_array_object GL_ARB_vertex_attrib_64bit GL_ARB_vertex_attrib_binding GL_ARB_vertex_buffer_object GL_ARB_vertex_program GL_ARB_vertex_shader GL_ARB_vertex_type_10f_11f_11f_rev GL_ARB_vertex_type_2_10_10_10_rev GL_ARB_viewport_array GL_ARB_window_pos GL_ATI_draw_buffers GL_ATI_texture_float GL_ATI_texture_mirror_once GL_S3_s3tc GL_EXT_texture_env_add GL_EXT_abgr GL_EXT_bgra GL_EXT_bindable_uniform GL_EXT_blend_color GL_EXT_blend_equation_separate GL_EXT_blend_func_separate GL_EXT_blend_minmax GL_EXT_blend_subtract GL_EXT_compiled_vertex_array GL_EXT_Cg_shader GL_EXT_depth_bounds_test GL_EXT_direct_state_access GL_EXT_draw_buffers2 GL_EXT_draw_instanced GL_EXT_draw_range_elements GL_EXT_fog_coord GL_EXT_framebuffer_blit GL_EXT_framebuffer_multisample GL_EXTX_framebuffer_mixed_formats GL_EXT_framebuffer_multisample_blit_scaled GL_EXT_framebuffer_object GL_EXT_framebuffer_sRGB GL_EXT_geometry_shader4 GL_EXT_gpu_program_parameters GL_EXT_gpu_shader4 GL_EXT_multi_draw_arrays GL_EXT_packed_depth_stencil GL_EXT_packed_float GL_EXT_packed_pixels GL_EXT_pixel_buffer_object GL_EXT_point_parameters GL_EXT_polygon_offset_clamp GL_EXT_provoking_vertex GL_EXT_rescale_normal GL_EXT_secondary_color GL_EXT_separate_shader_objects GL_EXT_separate_specular_color GL_EXT_shader_image_load_store GL_EXT_shader_integer_mix GL_EXT_shadow_funcs GL_EXT_stencil_two_side GL_EXT_stencil_wrap GL_EXT_texture3D GL_EXT_texture_array GL_EXT_texture_buffer_object GL_EXT_texture_compression_dxt1 G [... cut ...]
20:20:44.576 : OpenGL renderer:GeForce GTX 660 Ti/PCIe/SSE2
20:20:44.576 : Using fast VBOs for rendering is possible
20:21:14.081 : No start signal detected - forcing start
20:21:14.098 : N1 M110*34
20:21:14.098 : N2 M115*36
20:21:14.098 : N3 M105*36
20:21:14.098 : N4 M114*35
20:21:14.098 : N5 M111 S6*98
20:21:14.102 : N6 T0*60
20:21:14.102 : N7 M20*22
20:21:14.102 : N8 M80*19
20:21:14.102 : N9 M105*46
20:21:17.150 : N10 M105*22

The Board worked fine with Repetier Host before I upgraded the firmware.

TheSheepKing151, 2019-02-24:
The serial monitor from the arduino IDE is reading random ASCII characters.

repetier, 2019-02-25:
Assuming ack would be shown you see there are no responses from the printer.
Did you use right baud rate in arduino IDE? Default is 9600 baud I think, but you say it is on 115200 baud so you need to set that. If it was 115200 baud your printer is not set to that. Change eeprom mode to a different value != 0 and upload again to make sure it is set to what you said. Then communication should be readable.

TheSheepKing151, 2019-02-25:
Yes, 115200 was used in all programs and on the printer. The eeprom mode is 1 and baudrate set in configuration.h to 115200. Changing the baudrate in the printer settings doesn't solve the problem.

repetier, 2019-02-25:
That does not mean baud rate is set to 115200, therefore please do what I ask to make sure it is really set to 115200. So compile with eeprom mode 2 and upload.

TheSheepKing151, 2019-02-26:
Compiled with eeprom mode 2. Still same issue :/

qusai210, 2020-10-06:
hi 
i made the first 3d printer and the extruder does not work 
i used repetier firmware 1.0.4 with host 
can you help me reppetier

sandro730, 2020-12-28:
Hello
Delta Printer Firmware version 1.0.4
I have the same problem.
Even after using the M302 S1 command I cannot move the extruder to load the wire towards the hotend.

[Configuration.h](https://github.com/sandro730/Repetier-Firmware/blob/MyMaster/src/ArduinoAVR/Repetier/Configuration.h)

repetier, 2020-12-29:
Did you use small extrusions or a big one? Big ones are blocked normally. Big is defined here:
#define EXTRUDE_MAXLENGTH 100

which is less than you normally use for bowden.

Also check eeprom to see if all parameter like steps/acceleration etc are valid values.

sandro730, 2020-12-29:
Thanks I solved. 
It was enough to disable the automatic retraction.
In my experience, I recommend reporting the status of
the retraction feature, to the log when executing the M302 S1 command.

Thanks