repetier/RepetierFirmware #853: RH not handling Z offset

ruggb, 2019-02-25:
I thought this was a Marlin issue, but apparently not. Rather than posting this all over again, could you please look at
https://github.com/MarlinFirmware/Marlin/issues/13245

RH appears to be TRYING unsuccessfully to handle Zoffset for a 0.01 initial step, but not for 0.1 or 1.0.
thanks

repetier, 2019-02-25:
Ok, host has z min fixed to 0. Only reason you see -0.05 first is that it is from homing position. Moves will use the limits. SO I need to add a z min for the future. Have put it in my open list for host.

closed 2019-02-25 by repetier

ruggb, 2019-02-25:
Yes, the -0.05 is obviously what Marlin is reporting. Not sure I understand. If Zmin is fixed at 0 why does a 0.01 step go to 0? It would go to -0.04. Somehow you are looking at the initial position (-0.05) for a 0.01 step, but not for a 0.1 or, 1.0 step. It may only be fixed for a 0.1 and 1.0 step.
Zmin should be 0. But if the initial position reported is not 0 then the offset needs to be factored in for the 1st step after a home. Right?