repetier/RepetierFirmware #856: Running printer both coreXY and Cartesian use one board

AhmadSarifudin, 2019-03-21:
I have project use repetier firmware V2, and have plan to move my printer base use coreXY mode and Cartesian mode for my printer head, is it possible on repetier firmware v2?
my plan will use X and Y axis for base movement and A and B  axis for head movement  
or are there suggestion to make it work?

repetier, 2019-03-21:
What will be the logic here? Isn't one part enough?

Anyhow the answer is that you need your own PrinterType. Default corexy can have a and b as well but it will move xyz as core printer and ab as cartesian printer. Also ab only use default homing sequence which might be ok or not.

Since this is the opposite to what you want, you can copy the corexy and make your own type with some easy modifications. The .h/.cpp files are the only printer type dependent files. And just such cases are the reason for the isolation - to make it easy to add other types as well.

Can you explain what the aim of this double splitting is? Normally you control positions only with xyz and slicers will only output these coordinates.

If you want to use them both to double possible speed, there is an easy solution. There is already a core xyza solution build in. You can extend that easily to core xyzab and split 50% to ab and 50% to xy since you have full control over the factors.

skwhite16, 2019-03-21:
My impression was that there would 7 controllable axes. As it stands I am seeing 6 (XYZ,E,AB). Is there going to be a 7th axis? And if so would that also be a cartesian axis?

repetier, 2019-03-22:
7th axis is C axis yes. But you still need a logic to control them. So PrinterType defines the first axis to match the printer type which is why xyz are the corexyz axis. All remaining axis are always implemented as cartesian axis. Display will only show xyz axis. Other axis are considered helper axis whcih are of no interest for display, also you can of course modify the display.