repetier/RepetierFirmware #860: Repetier-firmware v2 error homing axis

AhmadSarifudin, 2019-04-11:
I have problem with homing axis , when sent Gcode G28 Y0 or G28 Z0  use Repetier host V2.1.3, I got homing Axis always on X axes , I use CoreXYZ printer type, are there there something wrong on my configuration?:
#define X_HOME_PRIORITY 0
#define Y_HOME_PRIORITY 1
#define Z_HOME_PRIORITY 2
// X motor = x + y
#define COREXYZ_X_X 1
#define COREXYZ_X_Y 1
#define COREXYZ_X_Z 0
// Y motor = x - y
#define COREXYZ_Y_X 1
#define COREXYZ_Y_Y -1
#define COREXYZ_Y_Z 0
// Z motor = z
#define COREXYZ_Z_X 0
#define COREXYZ_Z_Y 0
#define COREXYZ_Z_Z 1
//home dir
#define X_HOME_DIR 1
#define Y_HOME_DIR 1
#define Z_HOME_DIR -1

Thanks

repetier, 2019-04-11:
That is hard to say. For Z moves it can add x and y homing if FIXED_Z_HOME_POSITION is 1.

Pure Y homing should not trigger it. What is happening exactly? Does it home X plus wanted direction or does only X axis move?

If you move manually, do all axis move into correct direction? That would check the transformation matrix to be correct.

AhmadSarifudin, 2019-04-11:
yes , it does only X axis move , even if i tried move with on A axis use G28 A0 ,
yes,when I move manually , each step move into correct direction.
before this homing issue , I could not move all axis , and I checked on enable pin did not triggered to low state for enabling  the motor driver(A4988 motor driver), so I made changed on stepper.h
do it effect to the homing?

    inline void enable() final {
        enableCls::off(); // before on()
    }
    inline void disable() final {
        enableCls::on(); //before off()
    }