repetier/RepetierFirmware #864: Dual X Carriage with 4 Extruders

ssaid1, 2019-05-03:
Hi, 

      I am building 3d printer with dual x carriage. And each x extruder have 2 nozzles. Totally, there are two x axises and four nozzles. Can I use like this with repetier firmware without any problem? I could not do this in Marlin firmware ? Marlin firmware is introduce, there is just one nozzle in first x axis and other 3 of them are in second one. This should not be. Are there anyone who can help me about this ?

repetier, 2019-05-03:
With V1 firmware this will never be possible. For V2 version we have plans to support this, but it is not implemented at the moment. But being more flexible is will be much easier to do so here. I remember that I even have already added some structures to support this, but no final code - also due to fact that we have no such printer for testing.

ssaid1, 2019-05-06:
Thanks for your answer. I supposed, it is possible with small changes. If you add your firmware, I want to use it.

repetier, 2019-05-06:
Ok, so just watch V2 changes to see when it is there. Currently I'm more working on making different hardware work, then I guess I come back to such combinations.

ssaid1, 2019-05-06:
Take it easy.

closed 2019-05-06 by ssaid1